realsenseai/librealsense #5259: Changing output size of image

dedejoe, 2019-11-18:
Hi here's what I'm using:
| Camera Model                       | { D435i } | 
| Firmware Version                   | Current Firmware | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.19                                       | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            |  Current SDK                         | 
| Language                            |  Python                       | 
| Segment			|  AR                     | 

Since I'm using a raspberry pi 4 performance is key for me.  I want to run the resolution at 640x360, but draw the image at full screen, while still using the lower resolution depth data.  I can resize the image using openCV but thats more post processing than i want to do.  Is there an easy way in the SDK that I can use to upscale the image, while keeping the depth data resolution low?  In the same way that the real-sense viewer does.

Thanks!

dorodnic, 2019-11-18:
For rendering the Viewer (and all the examples) are creating opengl texture and uploading low-res depth data to the GPU. Then, its up to the GPU to stretch the texture to whatever polygon you wish to render to. This depends on your language and rendering subsystem, but I believe there is glfw variant for python as well.

RealSenseCustomerSupport, 2019-11-25:
----------------------------------------------

Hi @dedejoe,

Did @dorodnic's answer help to find a solution to your question?Please let us know if you need any further clarifications regarding this matter.

Thank you!

closed 2019-12-09 by RealSenseCustomerSupport