realsenseai/librealsense #5260: No more frame buffers (1), dropping frame

Makmarkrealsense, 2019-11-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {T265 } | 
| Firmware Version                   | 0.2.0.857 | 
| Operating System & Version | Linux raspberrypi 4.19.80-v8+  | 
| Kernel Version (Linux Only)    | 4.19.80-v8+                                         | 
| Platform                                 | Raspberry Pi4b  |
| SDK Version                            | 2.30.0                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {VR}                          | 

### Issue Description
I'm using POSE callback function in good lighting environment , it will happen run after 1 to 4 hrs and show the debug level message , no more POSE callback again :

10:10:21.853 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:21.855 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:21.886 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:21.890 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:21.919 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:21.922 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:21.953 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:21.955 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:21.986 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:21.989 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.019 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.022 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.053 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.056 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.086 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.089 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.119 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.123 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.153 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.156 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.186 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
10:10:22.190 [25686] [E] Device-6078: No more frame buffers (1), dropping frame
Killed
pi@raspberrypi:~/librealsense/build/examples/ar-basic $

jebasamuel, 2019-11-18:
I had faced similar issue. I believe it is due to frame buffer is not released as earlier possible when new frame is arrived. 
I think It may work well if we increase no of frame buffers from 1 to 2 or 3( double/triple frame buffer).

Makmarkrealsense, 2019-11-18:
@jebasamuel  
I see , but when I get current RS frame queue size it show ::
RS2_OPTION_FRAMES_QUEUE_SIZE :: 64 

it mean more then 1 or 3 , is that the frame queue size same as frame buffers ?

Makmarkrealsense, 2019-11-22:
I think this is c++ issue, so I close this thread

closed 2019-11-22 by Makmarkrealsense

jebasamuel, 2019-11-25:
@Makmarkrealsense 
Sorry for delayed response. frame queue size is different from frame buffer ( concurrent memory allocated in driver level ). Did you find the root cause of this?

Makmarkrealsense, 2019-11-26:
assume this is multi thread problem, the thread holding a variable cause pipe callback fail, when I change to single thread ,this error not happen again !!!