realsenseai/librealsense #5261: cannot get IMU data on a Mac OS platform

fanofpal, 2019-11-18:
Hello, recently I tried to develop on a D435i, unfortunately, I could not obtain IMU data(including accel and gyro) in realsense-viewer(Mac version), although RGB and depth images could be displayed normally. 
It could be sure that the D435i itself did not have any problems as the IMU data could be obtained normally on a windows10 platform.
Then I tried the API of python on the Mac platform, the codes are as follows:
`import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.accel)
config.enable_stream(rs.stream.gyro)
pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        accel = accel_data(frames[0].as_motion_frame().get_motion_data())
        gyro = gyro_data(frames[1].as_motion_frame().get_motion_data())
        print("accelerometer: ", accel)
        print("gyro: ", gyro)

finally:
    pipeline.stop()`
I got an error showed that 'RuntimeError: Frame didn't arrived within 5000'.  I suspected that there's no accel and gyro data in the stream.

dorodnic, 2019-11-18:
Hi @fanofpal 
This is a limitation of the SDK, documented in [release notes](https://github.com/IntelRealSense/librealsense/releases/tag/v2.30.0) and [support matrix document](https://github.com/IntelRealSense/librealsense/blob/development/doc/support-matrix.md).
The source of the issue is that Mac OS is preemptively locking any new HID device preventing Linux-style access. 
There is good progress re-writing IMU backend using Apple-specific APIs, but it is not complete yet.

fanofpal, 2019-11-19:
@dorodnic Thank you for the reply!

closed 2019-11-19 by fanofpal

MartyG-RealSense, 2021-02-15:
Hi @fanofpal   The new librealsense version **2.42.0** has introduced IMU support for MacOS.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0