realsenseai/librealsense #5262: IR frame format

daothanlong11, 2019-11-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |  Ubuntu 16.04 LTS | 
| Platform                                 | PC |
| SDK Version                            |  2.28.0                         | 
| Language                            |  C++                         | 

### Issue Description
This is my code for receive left ir camera, but it can't take ir frame with BGR8 or RGB8 format.
But in the Realsense-viewer, i am able to choose the BGR8 format for the ir camera and it can stream it. so i want to ask how can i set the IR camera with the BGR8 format. Thank for your support

My code:

rs2::pipeline pip;
rs2::config cfg;

cfg.enable_stream(RS2_STREAM_INFRARED,1,640,480,RS2_FORMAT_Y8);

pip.start(cfg);

g2-bernotas, 2019-11-20:
You can only select Y8 or Y16 for the IR camera as far as I am aware:
[image]

If you are talking about the colour camera, then:
``` C++
cfg.enable_stream(RS2_STREAM_COLOR,640,480,RS2_FORMAT_RGB8);
```

RealSenseCustomerSupport, 2019-11-26:
----------------------------------------------

g2-bernotas is correct that if you are using D435(i) series camera, Infrared 1 sensor only support Y8 or Y16 format.
For D415 series, Infrared 1 is capable of streaming RGB8 format.
https://www.intelrealsense.com/compare-depth-cameras/

Thank you.

closed 2019-12-06 by RealSenseCustomerSupport