realsenseai/librealsense #5267: Announcement:  Intel OneAPI for build, test and deploy across multiple hardware targets

MartyG-RealSense, 2019-11-18:
Hi everyone,

Intel has introduced its "write once, run anywhere" **OneAPI** toolkit, which has a base kit and add-on kits.

 https://software.intel.com/en-us/oneapi

closed 2019-11-26 by dorodnic

compiaffe, 2022-06-17:
@dorodnic are there plans to port the CUDA based stream alignment to oneAPI / SYCL? That sounds like a pathway to support stream alignment support on ARM devices without CUDA.

MartyG-RealSense, 2022-06-17:
Hi @compiaffe  The RealSense SDK 2.0 does not have any connections to OneAPI or SYCL.  From the time of its launch in late 2017, the SDK was designed with the philosophy of having minimal third-party integrations in its core in order to make it compatible with as wide a range of different hardware and software platforms as possible so that RealSense could be anywhere and everywhere.

The SDK already has an alternative graphics acceleration system called **GLSL Processing Blocks** that supports pointcloud, alignment and more and is 'vendor neutral ', meaning that it can work with any brand of GPU.  It is built into the SDK and can be enabled simply by changing rs2:: instructions (such as rs2:: align) to rs2::gl::

For example, rs2::align becomes rs2::gl::align, rs2:: pointcloud becomes rs2::gl::pointcloud, etc.

https://github.com/IntelRealSense/librealsense/pull/3654 has a very good pros and cons analysis of GLSL processing blocks and when it can be used.

There is also a C++ example program for a GLSL-equipped pointcloud called **rs-gl** here:

https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

compiaffe, 2022-06-19:
Cool, thanks for the pointer!