realsenseai/librealsense #5270: T265: A lot of USB problems on macOS

neilyoung, 2019-11-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.1.0.279
| Operating System & Version |  Ubuntu 16.04
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.2.9.0                           | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description
Well, I know, T265 on macOS is not _officially_ supported, but maybe one has a clue. I was able (finally) to compile under macOS Catalina. Tried the `rs-pose` sample first. It just runs very randomly.

Most the time I'm getting errors like these on attempt to start (sometimes it also breaks in the middle of a run):

```
~/Documents/librealsense/build/examples/pose $ ./rs-pose 
17:53:46.782 [0] [E] Device-E800: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
17:53:47.021 [0] [E] Device-E800: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
17:53:47.021 [0] [E] Device-E800: Entered state [ERROR_STATE]
^C

~/Documents/librealsense/build/examples/pose $ ./rs-pose 
17:54:01.217 [0] [E] Device-0000: MessageID 0x1006 (SLAM_6DOF_CONTROL) failed with status 0x8
17:54:01.217 [0] [E] Device-0000: Error Transferring SLAM_6DOF_CONTROL
17:54:01.217 [0] [E] Device-0000: Error: Failed setting 6dof Control (0x6)
RealSense error calling rs2_pipeline_start_with_config(pipe:0x7fe48470cbc0, config:0x7fe48470a980):
    Failed to start TM2 camera

```

Anybody having a clue what to do in order to make USB work better?

neilyoung, 2019-11-18:
@bfulkers-i  I can confirm your PR https://github.com/IntelRealSense/librealsense/pull/5213 works great on macOS Catalina. I hope it will get merged soon into development and or master.

Thanks

neilyoung, 2019-11-22:
@bfulkers-i I was applying the PR to master and I'm running into the same issues as described in #5276. Catch 22 now: Either have a working USB or a working python. Choose wisely...

bfulkers-i, 2019-11-27:
@neilyoung I believe I found the issue with the `development` branch and python and fixed them in #5336. For convenience I also currently have that patch on `t265_direct` #5213

closed 2019-12-11 by neilyoung