realsenseai/librealsense #5271: Adding wheel odometry input to T265

shannonwarren, 2019-11-18:
I have some concerns fusing wheel odometery data using the calibration file provided. 

I'm assuming my calibration file provided has the correct translation and rotation parameters however when I give a very low noise covariance of 0.000005 to the wheel odometer(just for testing purposes) hoping that the T265 tracking follows my wheel odometers; the tracking is completely off. 

Could it be because
1: My wheel odometry input in 10hz. 
2: Is the covariance value to low?

shannonwarren, 2019-11-20:
After poking around a bit, I've got my wheel odometry to work with the t265 with extremely low covariances values and an input frequency of 10hz. However, I still have some queries regarding the _calibration_odometry.json_ file.

1: I did not have to change the translation and rotation parameters in the calibration file even though my base_link frame and t265_ camera_pose frame are not aligned. I looked up the t265_realsense_node.cpp odometry callback function (realsense-ros package) and it seems that you're using transforming the odometry twist values to the t265 camera frame. 

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/t265_realsense_node.cpp#L63

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

Hi Shannon,

We've recently updated some documentation regarding wheeled odometry on the T265. Please see the "Appendix" section in link below and let us know if this helps clarify things.

https://github.com/IntelRealSense/librealsense/blob/development/doc/t265.md

Thanks

gglaspell, 2019-12-12:
I am thoroughly confused now. For a camera facing the same forward direction as the robot I had assumed W would be [-1.2092, 1.2092, 1.2092] as indicated in #3462, @shannonwarren it appears you have them listed as all 0's. Additionally the example here https://github.com/IntelRealSense/librealsense/blob/development/doc/t265.md isn't as helpful as I'd like since the robot frame has the exact as the camera frame so naturally W would be [0, 0, 0], but my robot frame has X as forward and Z up. When I attempt to include the wheel odometry from the robot into the t265 the map is jumping around, which wasn't an issue before trying to incoperate wheel odometry into the t265.

schmidtp1, 2019-12-12:
@gglaspell, @shannonwarren, please note that there is a difference between librealsense and the ROS wrapper.
The documentation above refers to the coordinate convention in librealsense, while @shannonwarren  seems to use the ROS wrapper ("t265_realsense_node.cpp"). As @shannonwarren noted, in ROS there is a coordinate transformation that aligns the output to ROS coordinate convention (https://www.ros.org/reps/rep-0103.html) and also tries to apply the same transformation/rotation to the wheel odometry input: https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/t265_realsense_node.cpp#L64. This has to be taken into account.

@shannonwarren, great you figured it out and got it to work, and sorry for the inconvenience! We are looking into improving the wheel odometry API.

shannonwarren, 2019-12-12:
@schmidtp1 Yes, I have the T265 running on an indoor robot with wheel odometry as the input. Thanks for the assistance. Will be doing some long term stability testing over the next couple of weeks.

I had a question regarding the ros wrapper:
1* Is there a plan to enable mapping; relocalization etc using the ros wrapper?

schmidtp1, 2019-12-19:
@shannonwarren, great!

The options are enabled on default. I assume you are asking about exposing the option to disable them? It should already be possible this way: https://github.com/IntelRealSense/realsense-ros/issues/923#issuecomment-531547065.

shannonwarren, 2020-01-07:
@schmidtp1 Got it, thanks. Had a few more questions. 

Currently, we set a static transform (extrinsic) between the camera and wheel odometry in the calibration odometry file. For future releases are you considering updating the extrinsic parameters during operation/motion?

Is there any way to zero out the 'z' values of odometry if working in a 2d environment without an external ROS node. Odometry tends to drift in z quite a bit over time.

gglaspell, 2020-01-27:
@schmidtp1 I am also experiencing some drift, even when the robot is perfectly stationary, when incorporating wheel odom into the t265 and wanted to verify I had calibration file setup correctly. Since I am using the **ROS wrapper**, I have **W** set to [0,0,0] as per your guidance. However, I am not sure if the wrapper corrects for **T** as well. I currently have **T** set in my JSON file as [0,-0.24,0.2], but the static transform I launch is args="0.2 0.0 0.24 0.0 0 0 /base_link /cam_t265_link 100". Based on the example I believe I have it set correctly.

RealSenseSupport, 2020-09-18:
Thank you for highlighting further updates, requirements and issues with wheel odometry on T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265.

closed 2020-09-21 by RealSenseSupport