realsenseai/librealsense #5273: USB2 mode auto exposure issue with firmware v5.11.15.0

oor001, 2019-11-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | v5.11.15.0 | 
| Operating System & Version |   Win 10) / Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    |  (e.g. 4.15.0)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.20.0 }                          | 
| Language                            |  C++                         | 
| Segment			|                      | 

### Issue Description
We have found an issue with the auto exposure of the IR cameras on the D435 module when using the latest firmware and a USB 2 connection.

We previously encountered issues with auto exposure (see https://github.com/IntelRealSense/librealsense/issues/4140 and https://support.intelrealsense.com/hc/en-us/community/posts/360033445713-D435-Sunlight-saturation-problem). This issue is now fixed but it seems that as a result of fixing the camera over-exposing, it is now underexposing, especially in dimly lit indoor environments. 

We have tested this with several cameras and firmware versions. There is no issue with USB 3 but unfortunately we have no option to use that. Here are some examples of the issue:
[image]
This image shows the issue - the IR image is very dim and the resulting depth map very sparse.
[image]
Alternately - using the previous firmware we see a well balanced IR image and dense depth map. (you can see the firmware update dialog indicating old firmware). 

As I mentioned, with USB 3 the image is very good quality - so we are wondering if there is some sort of bug that has been introduced while fixing the aforementioned issue with the USB2 version.

Thanks very much,

dime1991, 2019-11-19:
I am having the exact same problem with USB2

matthiaslexer, 2019-11-20:
Having the same problem with USB2, with USB3 it just works fine.

RealSenseCustomerSupport, 2019-11-26:
----------------------------------------------

Hi
Thanks for the feedback. Yes, that is known issue after FW5.11.15 released. It will be fixed in FW5.12.1.0 as planned.
Please stay tuned.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

closed 2019-12-03 by RealSenseCustomerSupport