realsenseai/librealsense #5274: make install error on T265 on Raspberry Pi 0: "config.h": No such file or directory

snowymo, 2019-11-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |  Raspbian | 
| Platform                                 | Raspberry Pi 0|
| SDK Version                            |  v2.29.9                          | 

### Issue Description
After I successfully ran ``` cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true ```, I ran ```sudo make install```. It reported that they failed to find ```config.h```. I found some [page ](https://sourceforge.net/p/lirc/tickets/250/) clarifying it is related to the OS itself, but I don't know where to apply this patch and if it is really relevant. Any clue?
```
pi@raspberrypi:~/Desktop/librealsense/build $ sudo make install
[  0%] Performing update step for 'libusb'
[  0%] Performing configure step for 'libusb'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Desktop/librealsense/build/libusb-prefix/src/libusb-build
[  0%] Performing build step for 'libusb'
Scanning dependencies of target usb
[  9%] Building C object CMakeFiles/usb.dir/libusb/core.c.o
/home/pi/Desktop/librealsense/build/third-party/libusb/libusb/core.c:23:10: fatal error: config.h: No such file or directory
 #include "config.h"
          ^~~~~~~~~~
compilation terminated.
make[5]: *** [CMakeFiles/usb.dir/build.make:63: CMakeFiles/usb.dir/libusb/core.c.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:73: CMakeFiles/usb.dir/all] Error 2
make[3]: *** [Makefile:130: all] Error 2
make[2]: *** [CMakeFiles/libusb.dir/build.make:112: libusb-prefix/src/libusb-stamp/libusb-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/libusb.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

snowymo, 2019-11-20:
I guess maybe it is because libusb so I installed that by
```sudo apt-get install libusb-1.0-0-dev```
Then I ran all the ```cmake``` , ```make``` and ```make install```. All successful.

Then I ran the example code in python wrapper and have this error
```
21:12:39.925 [866] [E] TrackingManager: Error while loading image - libusb_open failed (0xFFFFFFFD), will try again later
```

I think it is still related to libusb. But I did not find anything related to raspbian, T265 and libusb online.

snowymo, 2019-11-20:
I realized some [pages ](https://ubuntuforums.org/showthread.php?t=901891) talked about usb rules. So I tried adding one file ```99-realsense-libusb.rules``` in ```\etc\udev\rules.d```.

I can share the file content if necessary.

Before that, I tried installing a couple of packages. See [here](https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide).

Nothing else has been changed.

RealSenseCustomerSupport, 2020-01-03:
----------------------------------------------

Hi,

Are you still experiencing issues with this. A few updates have gone in that resolve usb rules disparities.

closed 2020-01-22 by RealSenseCustomerSupport