realsenseai/librealsense #5275: Installing pyrealsense2 on MacOS?

seanngpack, 2019-11-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Operating System & Version |   {MacOS  Catalina }                             | 

### Issue Description
Hello all, this is a pretty high level question but I cannot really find the answer to--how do I install pyrealsense2 on macOS? It seems like `pip install` is not supported on Mac, so I am reading I have to build Python bindings from the source. However, I already have librealsense2 installed using brew, would there be a conflict? It seems pretty confusing to me and it'd be helpful if I could get a clear pipeline of how to get pyrealsense2 on my Mac.

Thanks!

seanngpack, 2019-11-20:
Okay I ran alternative steps to get something built, but I'm not sure what to do next.

1. `git clone https://github.com/IntelRealSense/librealsense`
2. `mkdir build` & `cd build`
3. `cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true`
4. `make -j4`

It built...but what is next? How do I actually use the wrapper (pyrealsense2) in my projects?

seanngpack, 2019-11-20:
Okay I got it, for future Mac users this is how I did it:

closed 2019-11-20 by seanngpack

neilyoung, 2019-11-20:
- `sudo make install` after make
-  `export PYTHONPATH=$PYTHONPATH:/usr/local/lib` either on console or in your `.bashrc`
- `import pyrealsense2` from everywhere

ducciopiovani, 2019-11-27:
@neilyoung thank you, but what  if you are working on a virtual environment ?

neilyoung, 2019-11-27:
@ducciopiovani Not sure. Never tried that.

TheButlah, 2019-12-12:
Why is the mac version not published to PyPI if its officially supported?

ivorycirrus, 2019-12-16:
@ducciopiovani I tried this on MacOS Catalina, python 3.7, anaconda environment. 
Build librealsense2 and install, then make symbolic link of realsense sdk libraries into the python virtual environment.

In my case] env name : rs2 , python version : 3.7
```cd /opt/anaconda3/envs/rs2/lib/python3.7```
```ln -s /usr/local/lib/pyrealsense2.cpython-37m-darwin.so pyrealsense2.so```
```ln -s /usr/local/lib/pybackend2.cpython-37m-darwin.so pybackend2.so```

ddxue, 2020-03-16:
Hi -- thanks for the above instructions! I am also trying to run my D435i on Catalina. 

I updated the firmware of the D435i and ran the instructions above 
`
git clone https://github.com/IntelRealSense/librealsense
mkdir build & cd build
cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true
make -j4
`

However when I try to `import pyrealsense2` I get this error. Any thoughts?
`>>> import pyrealsense2 as rs
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dlopen(/usr/local/lib/pyrealsense2.so, 2): Library not loaded: @rpath/librealsense2.2.dylib
  Referenced from: /usr/local/lib/pyrealsense2.2.17.0.so
  Reason: image not found`

ddxue, 2020-03-16:
> Hi -- thanks for the above instructions! I am also trying to run my D435i on Catalina.
> 
> I updated the firmware of the D435i and ran the instructions above
> `git clone https://github.com/IntelRealSense/librealsense`
> `mkdir build & cd build`
> `cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true`
> `make -j4`
> 
> However when I try to `import pyrealsense2` in python3 I get this Segmentation error. Any thoughts?
> `[1]    61662 segmentation fault  python3`

seanngpack, 2020-03-16:
> `[1] 61662 segmentation fault python3`
Maybe you're not running `sudo make install` after the instructions -- I'll update my post because I might've skipped a step lol

breathedigital, 2020-03-22:
> Okay I got it, for future Mac users this is how I did it:
> 
> 1. `git clone https://github.com/IntelRealSense/librealsense`
> 2. `mkdir build` & `cd build`
> 3. `cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true`
> 4. `make -j4`
> 5. `sudo make install`
> 6. You should still be in the /build directory so now you want to find the .so files. `cd wrapper` & `cd python`
> 7. You'll see a bunch of .so files in here, just copy them and then paste them to the root project directory of your project that you want to use pyrealsense2 in. Then you can just import pyrealsense2 in your project modules using `import pyrealsense2`
> 
> thanks @neilyoung-- alternative to steps 6 and 7, `export PYTHONPATH=$PYTHONPATH:/usr/local/lib` or add that to your .bashrc

It works like a magic!! Thank you.

neilyoung, 2020-03-22:
I'm still fine with just the export statement in .bashrc on all of my systems (macOS, Raspbian, Ubuntu). That together with `sudo make install` after build does it.

whubaichuan, 2020-09-10:
@neilyoung Hi, how do you install librealsense-python in macos?

whubaichuan, 2020-09-10:
@neilyoung @ivorycirrus @dorodnic Hi, I got the error information: 'common/fw/target.h' file not found.

neilyoung, 2020-09-10:
> @neilyoung @ivorycirrus @dorodnic Hi, I got the error information: 'common/fw/target.h' file not found.

AFAIK there is still no PyPI for realsense. I downloaded the sources and compiled from scratch, with PYTHON dependencies. Works great.

whubaichuan, 2020-09-10:
@neilyoung The [link](https://github.com/IntelRealSense/librealsense/issues/6246#issuecomment-614695052) said that python 3.7 has the Pypi for realsense. What's more, is the error 'common/fw/target.h' file not found related with the internet? Because I encounter the "No internet connection, disabling IMPORT_DEPTH_CAM_FW".

neilyoung, 2020-09-10:
@whubaichuan Oh, really? That's new to me. I was explicitly excluding macOS from my requirements.txt 

```
# pyrealsense2 is currently only available for Windows and some Linux architectures
pyrealsense2==2.32.1.1299;sys_platform == 'win32' or (sys_platform == 'linux' and platform_machine == 'x86_64')
```

Reverted this now and added the latest PyPI to no avail:

```
ERROR: Could not find a version that satisfies the requirement pyrealsense2==2.38.1.2225 (from -r requirements.txt (line 13)) (from versions: none)
ERROR: No matching distribution found for pyrealsense2==2.38.1.2225 (from -r requirements.txt (line 13))
```

Command used: `pip3 install -r requirements.txt` on macOS

whubaichuan, 2020-09-10:
@neilyoung Hi, I got the following output when I make install. But I can not see the librealsense2.so and pyrealsense2.so. After I export PYTHONPATH=$PYTHONPATH:/usr/local/lib, I still can not import pyrealsense2.

[  1%] Built target fw
[  7%] Built target realsense-file
[ 44%] Built target realsense2
[ 48%] Built target pyrealsense2
[ 54%] Built target pybackend2
[ 58%] Built target realsense2-gl
[ 59%] Built target rs-hello-realsense
[ 59%] Built target rs-software-device
[ 59%] Built target rs-capture
[ 60%] Built target rs-callback
[ 60%] Built target rs-save-to-disk
[ 61%] Built target rs-multicam
[ 61%] Built target rs-pointcloud
[ 63%] Built target rs-align
[ 64%] Built target rs-align-advanced
[ 65%] Built target rs-sensor-control
[ 65%] Built target rs-measure
[ 65%] Built target rs-depth
[ 66%] Built target rs-color
[ 66%] Built target rs-distance
[ 67%] Built target rs-post-processing
[ 68%] Built target rs-record-playback
[ 69%] Built target rs-motion
[ 70%] Built target rs-pose
[ 71%] Built target rs-pose-predict
[ 71%] Built target rs-pose-and-image
[ 72%] Built target rs-trajectory
[ 72%] Built target rs-ar-basic
[ 73%] Built target rs-ar-advanced
[ 74%] Built target rs-tracking-and-depth
[ 74%] Built target rs-convert
[ 75%] Built target rs-enumerate-devices
[ 75%] Built target rs-fw-logger
[ 76%] Built target rs-terminal
[ 77%] Built target rs-record
[ 78%] Built target rs-fw-update
[ 79%] Built target rs-data-collect
[ 87%] Built target realsense-viewer
[ 96%] Built target rs-depth-quality
[ 99%] Built target rs-rosbag-inspector
[100%] Built target rs-benchmark
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/librealsense2.2.38.1.dylib
-- Up-to-date: /usr/local/lib/librealsense2.2.38.dylib
-- Up-to-date: /usr/local/lib/librealsense2.dylib
-- Up-to-date: /usr/local/include/librealsense2
-- Up-to-date: /usr/local/include/librealsense2/rs_advanced_mode.h
-- Up-to-date: /usr/local/include/librealsense2/rs.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_device.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_internal.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_processing.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_serializable_device.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_record_playback.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_pipeline.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_types.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_frame.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_context.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_sensor.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_export.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_options.hpp
-- Up-to-date: /usr/local/include/librealsense2/rs.h
-- Up-to-date: /usr/local/include/librealsense2/h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_frame.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_config.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_context.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_record_playback.h
[... 47 lines cut ...]
-- Installing: /usr/local/bin/rs-depth
-- Installing: /usr/local/bin/rs-color
-- Installing: /usr/local/bin/rs-distance
-- Installing: /usr/local/bin/rs-post-processing
-- Installing: /usr/local/bin/rs-record-playback
-- Installing: /usr/local/bin/rs-motion
-- Installing: /usr/local/bin/rs-pose
-- Installing: /usr/local/bin/rs-pose-predict
-- Installing: /usr/local/bin/rs-pose-and-image
-- Installing: /usr/local/bin/rs-trajectory
-- Installing: /usr/local/bin/rs-ar-basic
-- Installing: /usr/local/bin/rs-ar-advanced
-- Installing: /usr/local/bin/rs-tracking-and-depth
-- Installing: /usr/local/bin/rs-convert
-- Installing: /usr/local/bin/rs-enumerate-devices
-- Installing: /usr/local/bin/rs-fw-logger
-- Installing: /usr/local/bin/rs-terminal
-- Installing: /usr/local/bin/rs-record
-- Installing: /usr/local/bin/rs-fw-update
-- Installing: /usr/local/bin/rs-data-collect
-- Installing: /usr/local/bin/realsense-viewer
-- Installing: /usr/local/bin/rs-depth-quality
-- Installing: /usr/local/bin/rs-rosbag-inspector
-- Installing: /usr/local/bin/rs-benchmark
-- Installing: /usr/local/lib/libfw.a

whubaichuan, 2020-09-10:
@neilyoung  I make it.
But when I run my code, it prints 'pyrealsense2' has no attribute 'pipeline'. Do you have any ideas?

neilyoung, 2020-09-10:
No, since I don't know your code :)

jianchaoci, 2020-11-13:
steps I have followed, and it do works for me.

1. git clone https://github.com/IntelRealSense/librealsense ##### this will clone a librealsense file into you local file. you can find it at '/Users/cijianchao'
2. cd librealsense
3. mkdir build
4. cd build
5. cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true
6. make -j4
7. sudo make install
8. #####copy pybackend2.2.39.0.cpython-37m-darwin.so ## and ### pyrealsense2.2.39.0.cpython-37m-darwin.so ##
##into the python file of your virtual environment, eg- /Users/cijianchao/Applications/anaconda3/envs/capture/lib/python3.7
9. cd /Users/cijianchao/Applications/anaconda3/envs/capture/lib/python3.7
10. ln -s /usr/local/lib/pyrealsense2.2.39.0.cpython-37m-darwin.so pyrealsense2.so
      ln -s /usr/local/lib/pybackend2.2.39.0.cpython-37m-darwin.so pybackend2.so

note: two files pyrealsense2.2.39.0.cpython-37m-darwin.so pyrealsense2.so & pybackend2.2.39.0.cpython-37m-darwin.so pybackend2.so may be different in the latest version.

rebekkaBrtle, 2020-11-25:
Thanks for your help here! I am new to all of this especially on the Mac and what I was missing all the time was to type `python3` (im using 3.7) into the terminal and afterwards when ">>>" appears, type `import site; site.getsitepackages()` and afterwards `import pyrealsense2`. I did all the steps mentioned above before this, but I was missing this all along. Maybe its common knowledge for many but maybe I can help a few newbies like me

rajkundu, 2021-01-22:
I did the following, and it worked for me on macOS Catalina 10.15.7. Hope it helps anyone else!

1) Install homebrew if you don't already have it.
2) `brew install cmake libusb pkg-config`
3) `git clone https://github.com/IntelRealSense/librealsense` to clone librealsense
4) `cd librealsense` (or cd into wherever you cloned)
5) `mkdir build && cd build`
6) Find the path to your python executable. When I run `which python3`, I get `/Library/Frameworks/Python.framework/Versions/3.8/bin/python3`, so I'll use that in Step 7.
7) `cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_PYTHON_BINDINGS=true -G Unix\ Makefiles -DCMAKE_MACOSX_RPATH=ON -DCMAKE_INSTALL_RPATH="/usr/local/lib" -DPYTHON_EXECUTABLE=/Library/Frameworks/Python.framework/Versions/3.8/bin/python3` but replace the value of the `DPYTHON_EXECUTABLE` flag as appropriate for you (as you determined in Step 6)
8) `sudo make install` - I got a ton of output with warnings, but progressed (slowly, but steadily).
9) `cd wrappers/python`
10) `ls -l` and you'll see two .so files that are NOT symlinks (don't have ->). I saw the following output:
  ```
  total 63072
  drwxr-xr-x  7 raj   staff       224 Jan 22 17:06 CMakeFiles
  -rw-r--r--  1 raj   staff     48922 Jan 22 17:06 Makefile
  -rw-r--r--  1 raj   staff      9768 Jan 22 17:06 cmake_install.cmake
  -rwxr-xr-x  1 root  staff  14784952 Jan 22 17:22 pybackend2.2.41.0.cpython-38-darwin.so
  lrwxr-xr-x  1 root  staff        38 Jan 22 17:22 pybackend2.2.cpython-38-darwin.so -> pybackend2.2.41.0.cpython-38-darwin.so
  lrwxr-xr-x  1 root  staff        33 Jan 22 17:22 pybackend2.cpython-38-darwin.so -> pybackend2.2.cpython-38-darwin.so
  -rwxr-xr-x  1 root  staff  17436208 Jan 22 17:21 pyrealsense2.2.41.0.cpython-38-darwin.so
  lrwxr-xr-x  1 root  staff        40 Jan 22 17:21 pyrealsense2.2.41.cpython-38-darwin.so -> pyrealsense2.2.41.0.cpython-38-darwin.so
  lrwxr-xr-x  1 root  staff        38 Jan 22 17:21 pyrealsense2.cpython-38-darwin.so -> pyrealsense2.2.41.cpython-38-darwin.so
  -rw-r--r--  1 raj   staff      1221 Jan 22 17:06 pyrealsense2Config.cmake
  -rw-r--r--  1 raj   staff      1978 Jan 22 17:06 pyrealsense2ConfigVersion.cmake
  drwxr-xr-x  3 raj   staff        96 Jan 22 17:06 third_party
  ```
which told me that `pyrealsense2.2.41.0.cpython-38-darwin.so` and `pybackend2.2.41.0.cpython-38-darwin.so` were the two "real" .so files I was looking for.

11) Now, you can do many things, as others have said. However, I wanted to be able to `import pyrealsense2` universally (as if I had simply run `pip install pyrealsense2`. So, I copied the two files to my `site-packages` folder. I ran `python3 -m site --user-site` to get the path of this folder, which was `/Users/raj/Library/Python/3.8/lib/python/site-packages` (replace the `python3` command with the path to your python executable if necessary). HOWEVER, you also need to rename or symlink these files once copied to `site-packages` since they include version numbering. So, in the following `cp` commands, I copied and renamed in one command. I ran `cp pyrealsense2.2.41.0.cpython-38-darwin.so /Users/raj/Library/Python/3.8/lib/python/site-packages/pyrealsense2.cpython-38-darwin.so` for pyrealsense and `cp pybackend2.2.41.0.cpython-38-darwin.so /Users/raj/Library/Python/3.8/lib/python/site-packages/pybackend2.cpython-38-darwin.so` for pybackend.
12) Now I can fire up `python3` anywhere and `import pyrealsense2` seems to work!

worthlessFella, 2021-01-29:
import pyrealsense2.pyrealsense2 worked for me

EmersonDove, 2021-07-18:
I had an issue where `make -j4` threw an error where it couldn't find `OPENSSL_ROOT_DIR` on Mac. I solved by running:
`make -j4 -e OPENSSL_ROOT_DIR=/usr/local/opt/openssl`

art-niu, 2022-02-18:
> I did the following, and it worked for me on macOS Catalina 10.15.7. Hope it helps anyone else!
>
> 1. Install homebrew if you don't already have it.
> 2. `brew install cmake libusb pkg-config`
> 3. `git [...]

Thanks, it works for me on macOS Big Sur for realsense D455.
I changed step 7 so that people can just copy and paste to run it.
In step 11, the site_packages was point to different location, I added below lines to my .bashrc and .bash_profile 
`export PYTHONPATH=$PYTHONPATH:/usr/local/lib:/Library/Frameworks/Python.framework/Versions/3.10/bin/python3`

VKudlay, 2022-03-06:
Thanks so much for all of your help! I know the issue is closed, but I wanted to add my solution for my M1 system.

I was having issues getting this working on M1 (i.e. missing `-latomic`), but managed to get it to work by building through Xcode like in [this reference](https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk). I have a friend who will have to replicate this on his system soon, so I might revise this post if he has further issue. 

The following worked for me (M1 w/ MacOS Monterey 12.1 and Python 3.9): 

- From [rajkundu's previous post](https://github.com/IntelRealSense/librealsense/issues/5275#issuecomment-765743533) (which will be referred back to a lot): 

> 1. Install homebrew if you don't already have it.
> 2. `brew install cmake libusb pkg-config`
> 3. ` git clone https://github.com/IntelRealSense/librealsense` to clone librealsense
> 4. `cd librealsense` [...]

7. `brew install homebrew/core/glfw3` because the the other guide said so. Don't know if it matters.

8. `sudo xcode-select --reset`

9. Follow step 7 from rajkundu, but replace `-G Unix\ Makefiles` with `-G Xcode` to generate an Xcode project. To reiterate, set your python alias to the one you want to add for the package for and and run: 
```
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false \
    -DBUILD_PYTHON_BINDINGS=true -G Xcode -DCMAKE_MACOSX_RPATH=ON \
    -DCMAKE_INSTALL_RPATH="/usr/local/lib" -DPYTHON_EXECUTABLE=$(which python3)
```
If you get a missing threads issue, add the following arguments at the beginning of `CMakeCache.txt` ([source](https://stackoverflow.com/a/54606606/5003309)): 
```
CMAKE_THREAD_LIBS_INIT="-lpthread"
CMAKE_HAVE_THREADS_LIBRARY=1
CMAKE_USE_WIN32_THREADS_INIT=0
CMAKE_USE_PTHREADS_INIT=1
THREADS_PREFER_PTHREAD_FLAG=ON
```
10. Open the resulting `RealsenseWrappers.xcodeproj` file and hit the play button to build the project. Completion opens up the `rs-rosbag-inspector` application. The build will contain more unbuilt components. At the end of the day, the `.so` files that we want (see rajkundu's post) will be relatively located in `wrappers/python/Debug`. This chain might not be built yet, so follow the pathway and build the `.xcodeproj` file in the respective folders if you can't progress. Specifically, you'll have to build `wrappers/RealsenseWrappers.xcodeproj` and `wrappers/python/RealsensePythonWrappers.xcodeproj`.

11. Then follow 11 from rajkundu's post. In my case, the `site-packages` folder had not been instantiated, so the following worked for me with the desired `python` alias and the non-aliased `.so` files:
```
mkdir $(python3 -m site --user-site)
cp pybackend2.2.50.0.cpython-39-darwin.so $(python3 -m site --user-site)/pybackend2.cpython-39-darwin.so
cp pyrealsense2.2.50.0.cpython-39-darwin.so $(python3 -m site --user-site)/pyrealsense2.cpython-39-darwin.so
```
> 12. Now I can fire up `python3` anywhere and `import pyrealsense2` seems to work!

I have tested the resulting import and have been able to work with it so far. Hope this helps out in case the other methods don't work for you.

rajkundu, 2022-03-06:
@VKudlay Awesome! Thank you for sharing and keeping it going! :))

neilyoung, 2022-03-06:
@VKudlay Thanks for sharing, but I seem to miss something here.  I wasn't active for a long time here, but since I now also own a new Apple Silicon MBP 2021, I thought I could give it a shot again.

The CMake step produces this:

```
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_PYTHON_BINDINGS=true -G Xcode -DCMAKE_MACOSX_RPATH=ON -DCMAKE_INSTALL_RPATH="/usr/local/lib" -DPYTHON_EXECUTABLE=$(which python3)
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.50.0
-- Setting Unix configurations
-- using RS2_USE_LIBUVC_BACKEND
CMake Warning (dev) at /opt/homebrew/Cellar/cmake/3.22.3/share/cmake/Modules/ExternalProject.cmake:3503 (message):
  Policy CMP0114 is not set to NEW.  In order to support the Xcode "new build
  system", this project must be updated to set policy CMP0114 to NEW.

  Since CMake is generating for the Xcode "new build system",
  ExternalProject_Add will use policy CMP0114's NEW behavior anyway, but the
  generated build system may not match what the project intends.
Call Stack (most recent call first):
  CMakeLists.txt:7 (ExternalProject_Add)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /Users/decades/Documents/librealsense/build/external-projects/pybind11
objc[95798]: Class AMSupportURLConnectionDelegate is implemented in both /usr/lib/libamsupport.dylib (0x20ac93678) and /Library/Apple/System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x1081ec2c8). One of the two will be used. Which one is undefined.
objc[95798]: Class AMSupportURLSession is implemented in both /usr/lib/libamsupport.dylib (0x20ac936c8) and /Library/Apple/System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x1081ec318). One of the two will be used. Which one is undefined.
Command line invocation:
    /Applications/Xcode.app/Contents/Developer/usr/bin/xcodebuild -project pybind11-download.xcodeproj build -target ALL_BUILD -parallelizeTargets -configuration Debug -hideShellScriptEnvironment

User defaults from command line:
    HideShellScriptEnvironment = YES
    IDEPackageSupportUseBuiltinSCM = YES

note: Using new build system
note: Planning
Analyze workspace

Create build description
Build description signature: eb0f6e815203c37c5a7248fffe394c2b
Build description path: /Users/decades/Documents/librealsense/build/external-projects/pybind11/build/XCBuildData/eb0f6e815203c37c5a7248fffe394c2b-desc.xcbuild

note: Build preparation complete
note: Building targets in dependency order
PhaseScriptExecution Generate\ CMakeFiles/ZERO_CHECK /Users/decades/Documents/librealsense/build/external-projects/pybind11/pybind11-download.build/Debug/ZERO_CHECK.build/Script-E68D2CF2B590D5A5E6A449DC.sh (in target 'ZERO_CHECK' from project 'pybind11-download')
    cd /Users/decades/Documents/librealsense/build/external-projects/pybind11
    /bin/sh -c /Users/decades/Documents/librealsense/build/external-projects/pybind11/pybind11-download.build/Debug/ZERO_CHECK.build/Script-E68D2CF2B590D5A5E6A449DC.sh
make: `/Users/decades/Documents/librealsense/build/external-projects/pybind11/CMakeFiles/cmake.check_cache' is up to date.

PhaseScriptExecution Generate\ src/pybind11-stamp/$(CONFIGURATION)$(EFFECTIVE_PLATFORM_NAME)/pybind11-update /Users/decades/Documents/librealsense/build/external-projects/pybind11/pybind11-download.build/Debug/pybind11.build/Script-9F192CAED326AA53D8937BCB.sh (in target 'pybind11' from project 'pybind11-download')
    cd /Users/decades/Documents/librealsense/build/external-projects/pybind11
    /bin/sh -c /Users/decades/Documents/librealsense/build/external-projects/pybind11/pybind11-download.build/Debug/pybind11.build/Script-9F192CAED326AA53D8937BCB.sh

PhaseScriptExecution Generate\ src/pybind11-stamp/$(CONFIGURATION)$(EFFECTIVE_PLATFORM_NAME)/pybind11-patch /Users/decades/Documents/librealsense/build/external-projects/pybind11/pybind11-download.build/Debug/pybind11.build/Script-0651F091D0D31F6850CBD7B8.sh (in target 'pybind11' from project 'pybind11-download')
    cd /Users/decades/Documents/librealsense/build/external-projects/pybind11
    /bin/sh -c /Users/decades/Documents/librealsense/build/external-projects/pybind11/pybind11-download.build/Debug/pybind11.build/Script-0651F091D0D31F6850CBD7B8.sh

PhaseScriptExecution Generate\ src/pybind11-stamp/$(CONFIGURATION)$(EFFECTIVE_PLATFORM_NAME)/pybind11-configure /Users/decades/Documents/librealsense/build/external-projects/pybind11/pybind11-download.build/Debug/pybind11.build/Script-770D6F825ED7864CFF12183F.sh (in target 'pybind11' from project 'pybind11-download')
    cd /Users/decades/Documents/librealsense/build/external-projects/pybind11
    /bin/sh -c /Users/decades/Documents/librealsense/build/external-projects/pybind11/pybind11-download.build/Debug/pybind11.build/Script-770D6F825ED7864CFF12183F.sh

PhaseScriptExecution Generate\ src/pybind11-stamp/$(CONFIGURATION)$(EFFECTIVE_PLATFORM_NAME)/pybind11-build /Users/decades/Documents/librealsense/build/external-projects/pybind11/pybind11-download.build/Debug/pybind11.build/Script-7CA08AF98FE251A8544A43EE.sh (in target 'pybind11' from project 'pybind11-download')
    cd /Users/decades/Documents/librealsense/build/external-projects/pybind11
[... cut ...]
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for L53X_FW_Image-3.5.5.1.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/decades/Documents/librealsense/build
```

Trying to `sudo make install` after ends up in `make: *** No targets specified and no makefile found.  Stop.`

Opening the `RealSenseWrappers.xcodeproj` and hitting `Run` ends immediately with `Build complete` but nothing happened so far...

I also didn't get `missing threads issues`.

VKudlay, 2022-03-06:
@neilyoung Thanks for trying it out! I decided to re-run the process and will post the video of it as soon as it's done. On my end, I had a few more things come up in the cmake phase. Just making sure, run cmake from your build directory? Then, did you build `build/librealsense2.xcodeproj` and then `build/wrappers/RealsenseWrappers.xcodeproj` in that order? When you hit the play button, do you see a progress bar at the top of xcode such as below? 
[image]

neilyoung, 2022-03-06:
@VKudlay Thanks for the follow up.

> did you run cmake from your build directory?

Yes

> did you build build/librealsense2.xcodeproj 

Hmm no... Is that required? I didn't even open this project...

Will repeat too :) 

Thanks

VKudlay, 2022-03-06:
@neilyoung Awesome! Having re-ran everything, I modified my post to reflect what all I've seen. As you said, threads are no longer an issue (it was an issue when trying to build using normal install, but I guess Xcode takes care of it). I've also posted a video of me going through the process.

neilyoung, 2022-03-06:
@VKudlay Thanks again. I will go through the entire process again from scratch tomorrow and let you know.

neilyoung, 2022-03-06:
BTW: You can also drop step 6 from your first @rajkundu  quote. You later determine the Python3 location by using `which python3``

neilyoung, 2022-03-06:
I'm sorry, step 10 fails on my machine. Build is claiming to build 187 files and in the end it fails.

[image]

neilyoung, 2022-03-06:
OK, after the failing step 11 (I have ignored this for now) there was this:

```
librealsense/build/Debug $ tree
.
├── librealsense2.2.50.0.dylib
├── librealsense2.2.50.dylib -> librealsense2.2.50.0.dylib
└── librealsense2.dylib -> librealsense2.2.50.dylib

```

But nothing in `librealsense/build/wrappers/python/Debug`.

I then opened `librealsense/build/wrappers/RealsenseWrappers.xcodeproj` in XCode and compiled it. Success. Now I found this in `librealsense/build/wrappers/python/Debug`:

```
librealsense/build/wrappers/python/Debug $ tree
.
├── pybackend2.2.50.0.cpython-39-darwin.so
├── pybackend2.2.cpython-39-darwin.so -> pybackend2.2.50.0.cpython-39-darwin.so
└── pybackend2.cpython-39-darwin.so -> pybackend2.2.cpython-39-darwin.so

0 directories, 3 files
```

Not bad, but no `pyrealsense2.2.50.0.cpython-39-darwin.so`...

neilyoung, 2022-03-06:
Not sure if it's related, but the missing glfw references manifest like so:

```
ld: warning: directory not found for option '-L/usr/local/opt/openssl@1.1/lib'
ld: warning: ignoring file /usr/local/lib/libglfw3.a, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
```

neilyoung, 2022-03-06:
Good. I solved it. I know, that the following might be something special to my machine, but in case, I stumble again upon this, it is written down here (and maybe it helps others).

So my M1 is using a grown system (stable since 2008), having been installed via TimeMachine from an older Intel Mac. I already earlier solved the problem of replacing the X86_64 brew packages with the proper ARM versions. But for some reason I had a X86_64 `libglfw3.a` on my system (see above) and this was very persistent. It didn't disappear with `brew uninstall`. 

So doing a `find . -name "libglfw*" 2>/dev/null` followed by an `otool -hv ./usr/local/lib/libglfw3.a` revealed that old X86_64 lib, which was discovered by CMake and configured to be linked with the project (what failed, as seen above).

On the way to find this I discovered, that the old `brew install homebrew/core/glfw3` suggestion is deprecated and should be replaced by `brew install glfw`.

However, this installs glfw as dylib, not as static lib as required by librealsense CMake. Instead of trying to make that run, I ran `brew edit glfw` and changed `-DBUILD_SHARED_LIBS=TRUE` to `-DBUILD_SHARED_LIBS=FALSE`.  Followed by `brew reinstall --build-from-source glfw`.

Now I had a static lib, but somewhere in `/opt/homebrew/lib/libglfw3.a`. Still not resolvable by CMake. Maybe an `ldconfig` could have helped, but I didn't try. Instead I tried to symlink this, but this also didn't work.

In the end I copied it into the place, CMake expected: `cp /opt/homebrew/lib/libglfw3.a /usr/local/lib/libglfw3.a`

Now CMake came through, I compiled the librealsense2.xcodeproj (which now also magically went up to over 1000 files to compile) and it finished w/o any error. I also didn't have to compile the RealsenseWrapper again, it was all there as required in `/build/wrappers/python/Debug`.

```
wrappers/python/Debug $ tree
.
├── pybackend2.2.50.0.cpython-39-darwin.so
├── pybackend2.2.cpython-39-darwin.so -> pybackend2.2.50.0.cpython-39-darwin.so
├── pybackend2.cpython-39-darwin.so -> pybackend2.2.cpython-39-darwin.so
├── pyrealsense2.2.50.0.cpython-39-darwin.so
├── pyrealsense2.2.50.cpython-39-darwin.so -> pyrealsense2.2.50.0.cpython-39-darwin.so
└── pyrealsense2.cpython-39-darwin.so -> pyrealsense2.2.50.cpython-39-darwin.so

0 directories, 6 files
```

I finally ran the copy procedure 

```
cd wrappers/python/Debug
mkdir $(python3 -m site --user-site)
cp pybackend2.2.50.0.cpython-39-darwin.so $(python3 -m site --user-site)/pybackend2.cpython-39-darwin.so
cp pyrealsense2.2.50.0.cpython-39-darwin.so $(python3 -m site --user-site)/pyrealsense2.cpython-39-darwin.so
```

and boom...

```
Python 3.9.10 (main, Jan 15 2022, 11:40:53) 
[Clang 13.0.0 (clang-1300.0.29.3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2
>>> pyrealsense2.__version__
'2.50.0'
>>> 
```

VKudlay, 2022-03-06:
@neilyoung I'm glad you got it fixed! I would say that I'm sorry for the delay, but I also don't think I would have been able to figure that out.

neilyoung, 2022-03-06:
@VKudlay No problem at all. This was special. Your preparatory work was deciding. Thanks again

menynexi, 2022-08-22:
Hello I am ghabing problems with running the command `cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true` and get the following error. 

CMake Error: The source directory "/Users/manuelgutierrez/Documents/realSenseHack" does not appear to contain CMakeLists.txt

neilyoung, 2022-08-22:
md build
cd build
cmake .. bla bla bla