realsenseai/librealsense #5276: T265: macOS python problem

neilyoung, 2019-11-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.1.0.279
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.2.9.0                           | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description
While trying to run one of the python samples I'm getting this error on macOS:

```
~/Documents/librealsense/wrappers/python/examples $ python3 t265_example.py
Traceback (most recent call last):
  File "t265_example.py", line 11, in <module>
    import pyrealsense2 as rs
ImportError: dlopen(/usr/local/lib/pyrealsense2.cpython-37m-darwin.so, 2): Symbol not found: __ZN3rs211save_to_ply15OPTION_PLY_MESHE
  Referenced from: /usr/local/lib/pyrealsense2.cpython-37m-darwin.so
  Expected in: flat namespace
 in /usr/local/lib/pyrealsense2.cpython-37m-darwin.so

```

Does that sound familiar? Any fix available?

dorodnic, 2019-11-19:
I think there is version mismatch between one of the parts - the script, python so file and realsense2.so file. There were some changes related to save_to_ply in recent releases, and you may be still running the old version of the SDK.

neilyoung, 2019-11-19:
Hmm. It is the development branch, cloned yesterday. My "Reguired Info" is wrong. The librealsense version is 2.30.1

neilyoung, 2019-11-19:
Somebody having another idea?

neilyoung, 2019-11-19:
@dorodnic You were right. I was on the current development branch and because of the USB issues on my Mac https://github.com/IntelRealSense/librealsense/issues/5270 I merged PR #5213, which made my USB run. But as a side effect it killed python. Maybe an interesting info for the team.

neilyoung, 2019-11-22:
I can only insist: The current development branch has the a.m. issue with python. The import fails on the missing external.

Not the case with the master branch.

Please don't ignore.

bfulkers-i, 2019-11-27:
@neilyoung I believe I found the issue with the `development` branch and python and fixed them in #5336. For convenience I also currently have that patch on `t265_direct` #5213

neilyoung, 2019-11-27:
@bfulkers-i  Oh, cool. I will give it a try if I have some time. Many thanks.

neilyoung, 2019-11-27:
@bfulkers-i Perfect. Works. USB - check, Python - check :) Thank you

neilyoung, 2019-11-27:
@bfulkers-i  Hmm, but there is a new issue: Up to now I was able to do this:

```
translation = rs.vector()
rotation = rs.quaternion()
pose_sensor = pipe.get_active_profile().get_device().first_pose_sensor()

...

pose_sensor.get_static_node("test", translation, rotation)

```

This code works fine with python 3.7 on Ubuntu. 

On Mac, latest development branch, python 3.7.5  and PR #5336  and #5213 applied it now produces this:

```
get_static_node(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pose_sensor, guid: str) -> Tuple[bool, pyrealsense2.vector, pyrealsense2.quaternion]

Invoked with: <pyrealsense2.pose_sensor object at 0x119ba6bb0>

```
Strange enough: If I return to master, the same problem occurs... Not sure, how this can be, since it worked two weeks ago... Did the function change lately?

neilyoung, 2019-11-27:
@bfulkers-i Correction: Master works with that code. Just not the patched development branch

neilyoung, 2019-11-27:
Looks like the function signature has changed:

`result, translation, rotation = pose_sensor.get_static_node("test")
`
That works. Is this the new way to call it?

lramati, 2019-11-28:
Yes. The function was updated to have a more pythonic signature

neilyoung, 2019-11-28:
@lramati OK, thanks for the info. Is there a smart way to decide programmatically, which approach to use? I mean, except exception handling ? :) I'm switching constantly between development and master.

lramati, 2019-11-28:
I assume that it should be doable with python's introspection tools, but I've never used them before

neilyoung, 2019-11-28:
Yeah, in other languages I would know.... 

i```
f ([obj respondsToSelector:@selector(methodName:withEtc:)]) {
   [obj methodName:123 withEtc:456];
}
```

I hated ObjectiveC much, but I would love if there would be something similar and easy to use in python :)

closed 2019-12-11 by neilyoung