realsenseai/librealsense #5277: Python API for A-factor

Microset, 2019-11-19:
| Required Info                         |                                                                |
| Camera Model                       |  D435 } | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |  Ubuntu 16  | 
| Kernel Version (Linux Only)    |  (e.g. 4.15.0)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  v2.30 }                          | 
| Language                            |  python                         | 
| Segment			|  Robot                          | 

Good day!
After #4807 I double check A_factor for Librealsense SDK v.2.30 and try to read and write A_factor:

advnc_mode = rs.rs400_advanced_mode(self.profile.get_device())
advnc_mode.get_amp_factor()
print(advnc_mode.get_amp_factor())
advnc_mode.set_amp_factor(0.08)

TypeError: set_amp_factor(): incompatible function arguments. The following argument types are supported:
1. (self: pyrealsense2.rs400_advanced_mode, group: STAFactor) -> None

Can read but not write A_factor.
Please check in your side.

Best regards

lramati, 2019-11-28:
as indicated by the error message, you need to pass in a struct of type STAFactor.
the code you want is
```python
factor = rs.STAFactor()
factor.a_factor = 0.08
advnc_mode.set_amp_factor(factor)
```

Microset, 2019-11-28:
Thank you lramati. That's working.

closed 2019-11-28 by Microset