realsenseai/librealsense #5279: Trying to test with lesser landmarks

RajkiranVeldur, 2019-11-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |     Can we try running the face anti-spoofing (rs-face-dlib) with 5 landmarks instead of 68?                                                           |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | D436 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | Ubuntu 18.04
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 5.0.0-36-generic
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  | PC
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | CPP
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | others 

###  Can we try running the face anti-spoofing (rs-face-dlib) with 5 landmarks instead of 68?            
As I'm researching on the concepts of Spoof detection using intel Realsense, I would like to know if we can run the same code (rs-face-dlib) with lesser landmarks, instead of more landmarks? Thanks you.

dorodnic, 2019-11-19:
The sample code is provided for reference, you are encouraged to modify it to your needs.

RajkiranVeldur, 2019-11-20:
Hey @dorodnic, can you please explain whats the calculation behind find_depth_from?

dorodnic, 2019-11-21:
@maloel - could you please comment?

maloel, 2019-11-21:
Hmm the code states:

`/*
    Calculates the average depth for a range of two-dimentional points in face, such that:
        point(n) = face.part(n)
    and puts the result in *p_average_depth.

    Points for which no depth is available (is 0) are ignored and not factored into the average.

    Returns true if an average is available (at least one point has depth); false otherwise.
*/`

Given 5 points, for example, for which only 3 have depth (the other 2 fall on depth=0), the average depth from those 3 points will be returned.

So if those 5 points are locations of landmarks for the nose, find_depth_from will return the average depth for the nose.

maloel, 2019-11-21:
@RajkiranVeldur I did have a version of the code working with only 5 points.
You'll need to use a different model (the source code has a pointer to it, I think) and, of course, update the heuristics, drawing, etc. that expect 68 points right now.

RajkiranVeldur, 2019-11-21:
@maloel can you please share the code if you have one? It will help me save a lot of time. I appreciate your support. Thank you.

maloel, 2019-11-21:
@RajkiranVeldur sorry, I no longer have it. :(

RajkiranVeldur, 2019-11-21:
Oh! Okay. No worries. Will try from my end. Thank you.

closed 2019-11-21 by RajkiranVeldur