realsenseai/librealsense #5284: 3D transformation using T265 output

drapado, 2019-11-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 & D400 } | 
| Firmware Version                   | 2.3 | 
| Operating System & Version |   Linux  | 
| Kernel Version (Linux Only)    |  4.9                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.3                          | 
| Language                            |  python                       | 
| Segment			|  Robot                       | 

### Issue Description
Hi, I'm using a combination of depth camera (D435) with the tracking camera (T265). I would like to know if there is an existing function in the library which can convert (update position with respect to the camera) a 3D point (x,y,z) measured at t0 by the D435 by using the tracking parameters (translation and rotation in quaternions) that the T265 generates when moving from t0 to t1. If not, do you guys now another python library that does this?
Thanks!

MartyG-RealSense, 2019-11-20:
Intel has an article that explores combining a depth camera with the tracking of the T265.

https://www.intelrealsense.com/depth-and-tracking-combined-get-started/

It offers a link to a code example, though it is for ROS rather than Python.  It says:

**"For an example of depth and tracking integration, a ROS package can be used to generate a 2D occupancy map based on depth images and poses from depth and tracking cameras respectively. 
 Internally, it uses a 3D representation to transform point clouds (the common output of a depth camera) into a common reference frame, using the pose data from the T265 tracking camera, and to accumulate information over time.  The accumulated measurements are mapped with each voxel (or volumetric pixel) having a probability ranging from zero (free) to 100 (occupied).  This output can then be used for robotic or drone navigation.**

The ROS package mentioned can be found here:

https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping/occupancy

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

Hi drapado,

Was the information provided by MartyG below helpful? Is it safe to close this issue?

Thanks

closed 2019-12-06 by drapado