realsenseai/librealsense #5285: Pipeline pipe causing app to crash

aalarcon96, 2019-11-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   |{ 05.11.15.00 } | 
| Operating System & Version |   { MacOS } | 
| Platform                                 | PC|
| Language                            |  {C++ }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello, 

I am using the realsense libraries to create my own app. There seems to be a problem with the rs2::pipeline pipe in that sometimes it works fine and other times it makes my app crash. Has anyone else had issues with the pipeline and starting the pipe? I saw on pipeline.cpp that there is an unsafe way to start it?

Thank you in advance.

RealSenseCustomerSupport, 2019-12-12:
----------------------------------------------

@aalarcon96 Which librealsense version did you get this issue? And did you get such crash with SDK examples such as rs-capture https://github.com/IntelRealSense/librealsense/blob/master/examples/capture? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@aalarcon96 Didn't get response from your side for weeks. Will close the ticket at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-02-03 by RealSenseCustomerSupport