realsenseai/librealsense #5287: set the color scheme with the RS2_OPTION_COLOR_SCHEME

gjd2017, 2019-11-21:
The **rs_measure** sample program demonstrates code in its first lines to set the color scheme with the RS2_OPTION_COLOR_SCHEME instruction.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure#code-overview

_Originally posted by @MartyG-RealSense in https://github.com/IntelRealSense/librealsense/issues/5194#issuecomment-550208681_

How can I do it in python?

dorodnic, 2019-11-21:
https://github.com/IntelRealSense/librealsense/issues/3967

gjd2017, 2019-11-21:
@dorodnic Thks , OK.

closed 2019-11-21 by gjd2017

lokeshgs87, 2019-12-27:
Thanks MartyG-RealSense, could please help us with java code . As we are implementing with Java wrappers.

MartyG-RealSense, 2019-12-27:
@lokeshgs87   Java support in the SDK is provided by an unofficial community contributed wrapper:

https://github.com/edwinRNDR/librealsense/tree/master/wrappers/java

I was therefore unable to find a specific Java reference to set the door scheme.  Wiring with Java in librealsense is discussed in the link below.

https://github.com/IntelRealSense/librealsense/issues/1594