realsenseai/librealsense #5288: System identification error using T265

ahuizxc, 2019-11-21:
Hi, when I testing T265 camera, there is a odd problem. When the T265 camera first plug in, run `dmesg` I get 
```bash
[272035.043097] usb 3-1.1: USB disconnect, device number 8
[272035.387418] usb 4-1.1: new SuperSpeed USB device number 4 using xhci_hcd
[272035.408591] usb 4-1.1: New USB device found, idVendor=8087, idProduct=0b37
[272035.408596] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[272035.408598] usb 4-1.1: Product: Intel(R) RealSense(TM) Tracking Camera T265
```
in my terminal.
And then when I run ```realsense-viewer```
and ```dmesg```
the Usb port information changed to :
```bash
[272035.043097] usb 3-1.1: USB disconnect, device number 8
[272035.387418] usb 4-1.1: new SuperSpeed USB device number 4 using xhci_hcd
[272035.408591] usb 4-1.1: New USB device found, idVendor=8087, idProduct=0b37
[272035.408596] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[272035.408598] usb 4-1.1: Product: Intel(R) RealSense(TM) Tracking Camera T265
[272035.408600] usb 4-1.1: Manufacturer: Intel(R) Corporation
[272035.408601] usb 4-1.1: SerialNumber: 8454121111440000
```
I am very confused:)

dorodnic, 2019-11-21:
Hi @ahuizxc 
This is normal - T265 is based on Intel Movidius chip. By default it loads as Movidius boot-loader. Viewer (and rest of the SDK apps) identifies it and loads T265 firmware to it, and the whole unit reboots as Intel RealSense Tracking Camera T265.
In theory, you could load any other Movidius firmware (like the NCS) but we only provide support for the firmware included in the SDK.

ahuizxc, 2019-11-22:
> Hi @ahuizxc
> This is normal - T265 is based on Intel Movidius chip. By default it loads as Movidius boot-loader. Viewer (and rest of the SDK apps) identifies it and loads T265 firmware to it, and the [...]

fine thanks

closed 2019-11-22 by ahuizxc