realsenseai/librealsense #5290: Disable Auto Exposure Priority in code

FangjinhuaWang, 2019-11-21:
Hi. I am working on D435i to synchronize Depth camera and RGB. I know that I should disable Auto Exposure Priority control under RGB Camera. But I do not know how to implement this in code and I do not find any example in the source code. Does this look like the following codes:

auto sensor = pipe_profile.get_device().first<rs2::depth_sensor>();
sensor.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0);

Thanks a lot!

dorodnic, 2019-11-21:
Hi @FangjinhuaWang 
Try `auto sensor = pipe_profile.get_device().query_sensors()[1];`
`RS2_OPTION_AUTO_EXPOSURE_PRIORITY` is not on the depth sensor, but rather on RGB

FangjinhuaWang, 2019-11-21:
thanks for your quick reply. I will have a try soon

FangjinhuaWang, 2019-11-22:
T

closed 2019-11-22 by FangjinhuaWang