realsenseai/librealsense #5291: Cannot get imu data using frame callback using a D435i

fanofpal, 2019-11-21:
Hello, recently I tried to use frame callback to get data from a D435i realsense camera. Referring to the code [in this issue](https://github.com/IntelRealSense/librealsense/issues/5143), I successfully obtained color and depth data. However, when I tried to obtain the imu data, there seemed to be something wrong. The program did not report any error and could run, but the imu data is not printed.
Here is my code, I suspect that the functions to get accel and gyro data are not used correctly.

`
from __future__ import print_function
import sys
import pyrealsense2 as rs
import cv2
import numpy as np

from threading import Lock
frame_mutex = Lock()
frame_data = {"accel" : None,
              "gyro" : None,
              "timestamp" : None
              }

def callback(frame):
    global frame_data
    if frame.is_frameset():
        frameset = frame.as_frameset()
        accel = frameset[0].as_motion_frame().get_motion_data()
        gyro = frameset[1].as_motion_frame().get_motion_data()

        timestamp = frameset.get_timestamp()
        frame_mutex.acquire()
        frame_data["accel"] = accel
        frame_data["gyro"] = gyro
        frame_data["timestamp"] = timestamp
        frame_mutex.release()

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 250)
config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 400)
frame_mutex = Lock()
pipeline.start(config, callback)

try:
    WINDOW_TITLE = 'Realsense'
    cv2.namedWindow(WINDOW_TITLE, cv2.WINDOW_AUTOSIZE)
    while True:
        frame_mutex.acquire()
        valid = frame_data["accel"] is not None
        frame_mutex.release()

        if valid:
            frame_mutex.acquire()
            frame_copy = {"accel"  : frame_data["accel"].copy(),
                                      "gyro" : frame_data["gyro"].copy()}
            frame_mutex.release()

            print('accel:', frame_copy['accel'])
            print('gyro:', frame_copy['gyro'])

        key = cv2.waitKey(1)
        if key == ord('q'):
            break
finally:
    pipeline.stop()
`

ev-mp, 2019-12-09:
Hi @fanofpal , 
When using the pipe callbacks the frames are expected to be propagated immediately, so the assumption that the callback's input is always a frameset is not complete- you have to add the `else` case that will provide for the case with frames arriving separately
```python
def callback(frame):
    global frame_data
    if frame.is_frameset():
        frameset = frame.as_frameset()
        accel = frameset[0].as_motion_frame().get_motion_data()
        gyro = frameset[1].as_motion_frame().get_motion_data()

        timestamp = frameset.get_timestamp()
        frame_mutex.acquire()
        frame_data["accel"] = accel
        frame_data["gyro"] = gyro
        frame_data["timestamp"] = timestamp
        frame_mutex.release() 
  **else**
            .....
```
Please see `rs-motion` [example](https://github.com/IntelRealSense/librealsense/tree/master/examples/motion) for implementation details. 
Additionally, in case of `accel` and `gyro` do arrive as a frameset it is not correct to assume that `frameset[0]` is acceleration and `frameset[1]` gyro -  it is something that should be discovered dynamically (#4018, #4803).

RealSenseCustomerSupport, 2019-12-19:
----------------------------------------------

Hi @fanofpal,

Has the comment provided by @ev-mp helped to find an answer to the issue?Please let us know if you need any further help with this matter.

Thank you!

closed 2019-12-26 by RealSenseCustomerSupport