realsenseai/librealsense #5293: T265: Feature request list_static_nodes

neilyoung, 2019-11-21:
Would it be possible to add such a function? This would allow applications to list all static nodes found in a map in order to get_static_node() later on. The advantage: Given, a node naming follows a syntax rule, the particular naming of a static node must not be know by a using app. Say, maps have been pre-created and contain static nodes, and the using app does not know the particular names (since they are encoding some stuff). Having a list function followed by a get_static_node with the particular name would be extremely helpful.

neilyoung, 2019-11-21:
A better name would probably `enumerate_static_node_names` or so...

RealSenseSupport, 2020-09-21:
Thank you for highlighting this feature. We have moved our focus to our next generation of products and consequently, we will not be addressing any new feature in the T265.

closed 2020-10-05 by RealSenseSupport