realsenseai/librealsense #5295: Realsense D435 Camera - Save-To-Disk with JPG

gstewart9727, 2019-11-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu 18.04.3 LTS  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |                         | 
| Language                            |  C++                          | 
| Segment			|  others                           | 

### Issue Description
Hello,

I am using the intel realsense camera and using it to feed images to a jetson xavier image recognition application. The save-to-disk example provided by Intel shows how to save depth images to disk as pngs, but I need to save them as jpgs. I located the alternative to the int stbi_write_png command (int stbi_write_jpg) in the nothings/stb, but using this causes a segmentation fault when running. 

I am wondering if there is an existing way to save depth images from the realsense camera to disk as a JPG without having to save as a png and then convert to jpg. If conversion is an absolute must, and example of how to do so would be greatly appreciated.

I have also posted on the Intel forums (https://support.intelrealsense.com/hc/en-us/community/posts/360037586633-Realsense-D435-Camera-Save-To-Disk-with-JPG?page=1#community_comment_360009542474) but unfortunately could not find an answer.

ev-mp, 2019-11-22:
Librealsense uses a 3rd party library [stb](https://github.com/nothings/stb) to generate snapshots. The library offers [several output formats](https://github.com/IntelRealSense/librealsense/blob/master/third-party/stb_image_write.h#L47-L51) including jpeg:
>   int stbi_write_**png**(char const *filename, int w, int h, int comp, const void *data, int stride_in_bytes);
     int stbi_write_**bmp**(char const *filename, int w, int h, int comp, const void *data);
     int stbi_write_**tga**(char const *filename, int w, int h, int comp, const void *data);
     int stbi_write_**jpg**(char const *filename, int w, int h, int comp, const void *data, int quality);
     int stbi_write_**hdr**(char const *filename, int w, int h, int comp, const float *data);

Hence saving jpeg instead of png in `rs-save-to-disk` demo , for instance, can be achieved by replacing [stbi_write_png(...)](https://github.com/IntelRealSense/librealsense/blob/9ef4ed363eb98386c138960ba878746b35d7f327/examples/save-to-disk/rs-save-to-disk.cpp#L46) call with [stbi_write_jpg(...)](https://github.com/IntelRealSense/librealsense/blob/master/third-party/stb_image_write.h#L50) in the following section:
>// Write images to disk
  std::stringstream png_file;
  png_file << "rs-save-to-disk-output-" << vf.get_profile().stream_name() << ".png";
...
  stbi_write_~~png~~**jpg**(png_file.str().c_str(), vf.get_width(), vf.get_height(),
          vf.get_bytes_per_pixel(), vf.get_data(), ~~vf.get_stride_in_bytes()~~ **int quality**);
...

gstewart9727, 2019-11-22:
> Librealsense uses a 3rd party library [stb](https://github.com/nothings/stb) to generate snapshots. The library offers [several output [...]

Please read the entirety of my post. I had mentioned that I have tried exactly that but it resulted in a segmentation fault. If you know why it faulted, please let me know.

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

Hi @gstewart9727,

Please let us know if you still need an assistance with this matter. If you do, please provide us with
a following information about the issue you have mentioned:
toolchain configuration
your code (where the segmentation fault occurs)
call stack
errors and segmentation faults messages/exceptions
logs

Thank you!

closed 2019-12-24 by RealSenseCustomerSupport