realsenseai/librealsense #5296: Serializing / Pickling frames / frame sets for remote align

compscidr, 2019-11-21:
I've read through the examples and documentation but can't seem to find a way that this is easily possible, so looking for some suggestions. We are using D435 and D415 cameras. Typically there are around 4 cameras. We're making use of 1920x1080x30fps RGB and 1280x720x30fps depth frames.

Each camera is attached to a small low power computer, and then being streamed back to a higher power computer over a very fast network. If we send back RGB and unaligned depth frames we can get excellent FPS at the central computer. However, if we do the align depth to color on the smaller, low power computers because they have slower CPUs, frames need to be dropped (it takes about 0.3s to do the frame align and send the frame to the central computer). If I move the camera to the higher power computer, and use CUDA, the frame alignment is quite fast (sending over localhost + doing the align takes about 0.05s). Unfortunately this is not a solution because the cameras need to be on things that move around and the wires come unplugged too often. 

The smaller computers do not have CUDA GPUs. I was wondering, if I might be able to send the frames to the central computer and do the align operation on it, rather than sending just the data from the frames. However, trying to simply pickle the frames is not possible. Any suggestions for how this might be accomplished? Currently I'm only sending the data of the frames. Is it possible to do an align operation with just this? Or is there some subset of other frame data I also need to do the operation?

The small computers are running ubuntu 18.04 and the more powerful computer is running ubuntu 16.04.

Running the master branch of pyrealsense2 compiled from GitHub. On the larger computer, compiled with CUDA support.

MartyG-RealSense, 2019-11-21:
A vendor-neutral alternative to CUDA in Librealsense is GLSL processing blocks.

https://github.com/IntelRealSense/librealsense/pull/3654

dorodnic, 2019-11-25:
One way to approach it is using built-in recorder - if you need to perform computation offline you can record stream to bag format (see playback & record example). Bag format preserves all information needed to perform alignment. Also, if you include `rs_export.hpp` you can use single frameset recorder to save each individual frameset to a bag file of its own. This will serialize frames + all implicit information required for alignment. 
Another approach is using ROS - [using-3-cameras-in-2-machines](https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines)
Also, there is [client server python sample](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server)

compscidr, 2019-11-25:
> One way to approach it is using built-in recorder - if you need to perform computation offline you can record stream to bag format (see playback & record example). Bag format preserves all [...]

Ah thanks, I'll give that a try - sounds promising. 

WRT to the GLSL blocks - it seems like I've compiled that in and its still not performing too well, so its likely the lower powered device has pretty low-end GPUs - its some type of integrated thing.

compscidr, 2019-11-26:
Turns out the writing to the file and reading it back in are quite slow, it's actually much faster to just align on the slower devices. I managed to disable the lz4 compression, which made it faster, but still not quite good enough. For reference, it's taking about 0.4-0.6 seconds to read the bag file and 0.1-0.3s to write it (and this is without even trying to send the bag file between the devices - which would add on another 0.06s or so - so total about 1s). On the slower device I was getting around 0.1s for the align, and that was producing about 20MB of data - so about 0.15s to transmit. In totally it was somewhere around 0.3s total.

I wonder how much of that is the disk i/o though and not just the serialization. In any case I think by digging to how the `src/media/ros/ros_writer.cpp` and some of the other classes in the cpp I could figure out serializing directly to the network to experiment since it seems like the bag stuff requires writing to a file. I have a feeling that a more specific serialization than the rosbag will save some time as well.

In order for this approach to make any sense, it will need to be possible for the CUDA align on the faster computer and the serialization and transmission to be < 0.3s. The CUDA align on the fast computer is currently taking about 0.04-0.05s, so that means I have ~0.25 for serialization and transmission on a 1Gbps link. The file size of the bag with just 1080p RGB and the highest resolution depth is around 7.7MB, so at 1Gbps the best I can do is 0.06, which leaves me around 0.2s for serialization/deserialization. Not sure if that is realistic or not.

RealSenseCustomerSupport, 2020-02-06:
----------------------------------------------

@compscidr Another possible way is to do timestamp synchronization for temporal alignment on the slower devices and then send to the paired frames to the faster computer to do space alignment.

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@compscidr Did you get chance to try? Any other questions? Looking forward to your update. Thanks!

compscidr, 2020-02-19:
Will try your suggestion within the next couple of days, just trying to make some time for it - thanks!

RealSenseCustomerSupport, 2020-03-06:
----------------------------------------------

@compscidr Could you please update if you still need support on this? Thanks!

compscidr, 2020-03-06:
Haven't had time to try it yet, but I think I have enough information to proceed. I'll close the issue and re-open if I have any problems.

closed 2020-03-06 by compscidr