realsenseai/librealsense #5297: Unity crashes if RealSense is unplugged while streaming

tnelling-draper, 2019-11-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.857 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.29 | 
| Language                            | C#/Unity | 
| Segment			|  I don't know what this means | 

### Issue Description
Steps to reproduce:
1. Plug RealSense device into PC
2. Open the Unity Editor
3. Add an RsDevice and RsPoseStreamTransformer
4. Hit 'Play' in the Editor
5. Unplug the RealSense
6. The following error may be reported:
```
ExternalException: rs2_pipeline_wait_for_frames(pipe:0000017F41F69EA0)
Rethrow as Exception: Frame didn't arrived within 5000
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <8f69383e07bb4010b4a8322e1cf5f071>:0)
Intel.RealSense.Pipeline.WaitForFrames (System.UInt32 timeout_ms) (at <8f69383e07bb4010b4a8322e1cf5f071>:0)
Intel.RealSense.SDK.RsDevice.WaitForFrames () (at C:/<path>/RsDevice.cs:159)
System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) (at <599589bf4ce248909b8a14cbe4a2034e>:0)
System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <599589bf4ce248909b8a14cbe4a2034e>:0)
System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <599589bf4ce248909b8a14cbe4a2034e>:0)
System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) (at <599589bf4ce248909b8a14cbe4a2034e>:0)
System.Threading.ThreadHelper.ThreadStart () (at <599589bf4ce248909b8a14cbe4a2034e>:0)
UnityEngine.UnhandledExceptionHandler:<RegisterUECatcher>m__0(Object, UnhandledExceptionEventArgs)
Intel.RealSense.NativeMethods:rs2_pipeline_wait_for_frames(IntPtr, UInt32, Object&)
Intel.RealSense.Pipeline:WaitForFrames(UInt32)
Intel.RealSense.SDK.RsDevice:WaitForFrames() (at C:/<path>/RsDevice.cs:159)
System.Threading.ThreadHelper:ThreadStart()
```
7. Unity Editor will eventually crash (you may need to stop playing) with the following error:
```
realsense2.dll caused an Access Violation (0xc0000005)
  in module realsense2.dll at 0033:1e464e1f.
```

This is reproducible on a regular Windows exe build as well. I haven't tested on other platforms.

wodndb, 2019-11-26:
I handling this external exception using try-catch block.

tnelling-draper, 2019-11-26:
> I handling external exception using try-catch block. But I don't know how to solve dll crush...

I couldn't figure out anything to put a try-catch block around, because the exception is being raised in a separate thread. Do you have a code snippet for your try-catch that you could share?

wodndb, 2019-11-26:
I used unitythread mode, so I didn't needed try-catch block.

If you want to use multithread mode of RealSense SDK, modify WaitForFrames() in RsDevice.cs

<RsDevice.cs>

```
    /// <summary>
    /// Worker Thread for multithreaded operations
    /// </summary>
    private void WaitForFrames()
    {
        while (!stopEvent.WaitOne(0))
        {
            try
            {
                using (var frames = m_pipeline.WaitForFrames())
                    RaiseSampleEvent(frames);
            }
            catch (Exception e)
            {
                Debug.Log(e);
            }
        }
    }
```

tnelling-draper, 2019-11-26:
Ah yes, come to think of it I did try that, and was able to successfully log the exception... but Unity still crashes.

wodndb, 2019-11-26:
So.. You means that unity is stop or down when RealSense camera is unplugged although exception is caught in try-catch block?

tnelling-draper, 2019-11-26:
> So.. You means that unity is stop or down when RealSense camera is unplugged although exception is caught in try-catch block?

If I run in the Editor, Unity will generally crash when I try to stop playing, even if I catch that exception. I was doing my testing of modifications to the RealSense SDK wrapper scripts in the Editor, so I didn't see whether Unity would still crash on a Windows build. That's something I'll look into the next time I have a chance to work on this.

wodndb, 2019-11-26:
In my case, when I unplug RealSense, streaming is stop. However unity don't stop or down. It works well. And Stop playing also works well.

I'll try to test that using current RealSense SDK for Unity.

RealSenseSupport, 2020-09-17:
Thank you for highlighting an issue with Unity crashes with T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-21 by RealSenseSupport