realsenseai/librealsense #5299: About the noise of SR305

sho-ji, 2019-11-22:
Noise occurs when the SR305 Coded-Light Depth Sensor is turned on.
Does noise occur in the Coded-Light pattern switching cycle?

MartyG-RealSense, 2019-11-22:
The coded light technology is vulnerable to being affected by nearby IR sources such as other SR300 / SR305 or by other cameras / devices using IR.  One example would be an external SR300 conflicting with a built in IR camera in a laptop.  I recall a past example of interference from Christmas lights too.  Do you have any nearby devices that might be emitting such interference please?

sho-ji, 2019-11-22:
Noise is sound, not interference.
The frequency analysis of noise has a peak at 5577.1Hz.

MartyG-RealSense, 2019-11-22:
The SR300 and SR305 cased USB cameras have a dual-array microphone.  There were a small number of past cases of the SR300 mic making an audible noise when placed near an electromagnetic source such as an audio speaker.

sho-ji, 2019-11-22:
There is no electromagnetic source near SR305.
Is the Coded-Light pattern cycle irrelevant?
Can you tell me about the Coded-Light pattern cycle?

MartyG-RealSense, 2019-11-22:
I do not have any information about what else other than EM interference could cause sound noise to emit from the SR30x microphone.

Information about the coded light pattern can be found in the links below.  The first couple are an introductory level and the third is more scientific.

https://www.youtube.com/watch?v=PluL7WTlKrM

https://software.intel.com/en-us/forums/realsense/topic/642822

https://vista.cs.technion.ac.il/wp-content/uploads/2019/05/IVCAMPAMI19.pdf

MartyG-RealSense, 2019-11-22:
It occurs to me that a poorly shielded USB cable may be a source of EM interference.  Are you using the official short one supplied with the camera or your own choice of cable, please?

sho-ji, 2019-11-22:
I use the official cable.

MartyG-RealSense, 2019-11-22:
Okay, thanks.  Hopefully one of the Intel team members on this forum will be able to provide advice for your problem, as they have access to more information than I do (I am not an Intel employee).  Good luck!

agrunnet, 2019-11-22:
The sr305 has a small MEMs mirror is that is scanning back and forth. I think this is what you are hearing.

sho-ji, 2019-11-25:
That's right. Do you have information about the scanning speed of the MEMs mirror?

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

Hi @sho-ji,

What's you going to handle this audible noise?

agrunnet, 2019-12-04:
It’s about 300hz frame rate.

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

Hi @sho-ji,

I'm going to close this thread out now since this is expected behavior.

closed 2020-01-09 by RealSenseCustomerSupport