realsenseai/librealsense #5302: uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

AlexGit2, 2019-11-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415 and D435  | 
| Firmware Version                   | RealsenseViewer: 2.30.0 Cameras: 05.11.15.00| 
| Operating System & Version |   Linux Ubuntu 18.04 LTS  | 
| Kernel Version (Linux Only)    |  5.0.0-36-generic                                         | 
| Platform                                 | PC: CPU Threadripper 1950x  |
| SDK Version                            |  SDK 2.0                       | 
| Language                            |                         | 
| Segment			|                           | 

I am working on a pc with Ubuntu 18.04 LTS (Kernel 5.0.0-36). I use the realsense viewer to display the right and left infrared image of the realsense d415 camera. At first glance everything seems to work perfectly. The camera is recognized and I can stream the left and right infrared image without problems. But if you look closely you notice that the right and left camera images are interchanged now and then for a short time (i think for the time of one frame). Whenever that happens, the frame rate, which is being displayed in the realsense viewer, drop to zero for a short time . After I noticed that, i looked at the dmesg output (terminal). Whenever the streams in the realsense viewer flicker, the following message was on dmesg: 
[ 2625.008920] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

I installed the librealsense software with the help of the manual (Linux Ubuntu Installation) https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

In another manual I have read, that after patching the kernel module uvcvideo with the script "./scripts/patch-realsense-ubuntu-lts.sh", the version name of the uvcvideo modul must include the string "realsenese". This is not the case for me. 
Here is the output that gives me the command "modinfo uvcvideo | grep "version:"":
version:        1.1.1
srcversion:     1B3F01A2600CA3708971F1B

Can it be that the module is not patched properly and therefore these errors occur?

I've tried the whole again with a D430 camea. Here too, exactly the same happens.

Is there anyone else who also has this problem and know how to solve it? 
Thanks in advance.

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

Hi @AlexGit2,

1. Is the issue of interchanged images and frame drop reproduced every time?
2. How about other librealsense apps like rs-capture? The same stream issue happens?
3. Any dmesg when you applied the patch? You can refer to #5231 to see if it's helpful about the patching.

mpowelson, 2019-12-13:
This doesn't really seem to be hindering me, but I am also experiencing this issue. I am running Ubuntu 18.04.3 (Kernel 5.0.0-37-generic) Several times a second it appears that they interchange. It doesn't look like it's actually effecting the depth image. I am also getting that error on dmesg, but it appears to only be on startup for me. I am however getting this error every few seconds, but the interchaning streams happens much more frequently 
```
 13/12 14:12:17,446 WARNING [140297069631232] (types.cpp:49) Out of frame resources!
 13/12 14:12:17,446 ERROR [140297069631232] (synthetic-stream.cpp:47) Exception was thrown during user processing callback!
```

After patching ( which said patched kernels modules were created successfully), my modinfo uvcvideo says

```
version:        1.1.1
srcversion:     4FA7FAEF8FFC0A9AB1BAC31```

RealSenseSupport, 2020-09-26:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

enricomrk, 2020-10-06:
Hi there,
I'm having the exact same dmesg output as the title of the issue.
It appears when my application starts, but also while running (even if not that often).
Would be nice to investigate the issue.

I'm on Ubuntu 18.04.5 updated today, and I'm using librealsense2 from Intel repositories.
Kernel is Linux nuci5 4.15.0-118-generic.
Modinfo uvcvideo says version is 1.1.2.realsense-1.3.14

I'm glad to provide other info if needed!

stfuchs, 2020-10-10:
same here:

Ubuntu 18.04.5 LTS
4.15.0-1096-oem
1.1.2.realsense-1.3.14
Intel RealSense D435 750612070465 05.12.07.100 

I'm occasionally experiencing frame drops but I'm not sure if it is related. So far I have been simply ignoring the dmesg errors, but would be nice to get it addressed some day.

lucbettaieb, 2020-10-16:
I am also experiencing lots of spam in dmesg on multiple kernel versions.
18.04.1-Ubuntu
5.4.0-51-generic and 5.3.0-62-generic
Intel RealSense D435I
05.12.05.00

P.S. Hi @stfuchs :wave:

doisyg, 2020-11-14:
Same dmesg spam here on Ubuntu kernel 5.4 (patched)

sinback, 2020-11-17:
same here (and I've seen it before as well with D435s)

Ubuntu 18.04
4.15.0-123-generic
Intel RealSense L515
01.05.01.03

ayehud, 2020-12-22:
same issue here

emeyedeejay, 2020-12-27:
+1 on an Rpi 3B+ running Home Assistant.

htawadegetample, 2021-03-18:
same issue with l515 intel realsense

htawadegetample, 2021-03-19:
same issue here. Even with l515 camera

RealSenseSupport, 2021-03-21:
Hi,

We are unable to reproduce this issue on our setup here. Do you experience the same results if you don't use kernel patching; ex: install the SDK via deb packages? What about if you build with the FORCE_RSUSB_BACKEND instead?

Thanks

Moore123, 2021-04-02:
On my desktop, Ubuntu-18.04LTS kernel 4.15.0-124-generic  with USB webcam:: 1b3f:2002

Will flooding  disk /var/log/uvcdynctrl-udev.log
tons of messages [ "uvcvideo: Failed to query (GET_INFO) UVC control 2 on unit 1: -32 (exp. 1).  ]

update to kernel 5.4.0-70-generic #78~18.04.1-Ubuntu SMP, the flooding stopped
somewhere  should be fixed in kernel-module  uvcvide.ko

RealSenseSupport, 2021-04-25:
Hi @AlexGit2 

Will you be needing further help with this?

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

howiechiangRTR, 2021-04-26:
I encountered this on ubuntu 18.04 kernel `4.15.0-142-generic`

RealSenseSupport, 2021-04-26:
Ok thanks we will investigate.

Thanks

parasharrk, 2021-05-13:
I am facing the same problem in `5.4.0-72-generic #80~18.04.1-Ubuntu SMP Mon Apr 12 23:26:25 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

ecmnet, 2021-05-15:
Same here. Connected both T265 and D455

gsaponaro, 2021-06-01:
I have the same issue.
Kernel: `5.4.0-73-generic #82~18.04.1-Ubuntu`.
Sensor information and error:

```
[ 9508.392742] usb 2-1: Product: Intel(R) RealSense(TM) Depth Camera 415 
[ 9508.392745] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 415 
[ 9508.392748] usb 2-1: SerialNumber: [omitted]
[ 9508.395659] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[ 9508.397689] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/input/input49
[ 9508.398516] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
...
[ 9860.154129] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 9860.253239] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 9860.372241] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 9860.499457] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[ 9860.507094] uvcvideo: Failed to query (SET_CUR) UVC control 10 on unit 2: -32 (exp. 2).
[ 9860.518723] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[ 9860.526261] uvcvideo: Failed to query (SET_CUR) UVC control 10 on unit 2: -32 (exp. 2).
[ 9863.768822] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
etc.
```

RealSenseSupport, 2021-06-18:
These types of errors are almost always related to needing correct kernel patches. Another option is to bypass altogether and use the RSUSB_BACKEND.

MartyG-RealSense, 2021-09-11:
Hi everyone, does anybody who commented on this case require further assistance with it, please?  Thanks!

MartyG-RealSense, 2021-09-17:
Case closed due to no further comments received.

closed 2021-09-17 by MartyG-RealSense

russellkir, 2022-04-11:
Hi @RealSenseSupport @MartyG-RealSense , I am experiencing similar issues:
```
[ 4694.898053] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 4694.988812] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 4695.307178] uvcvideo: Failed to query (GET_CUR) UVC control 2 on unit 3: -32 (exp. 1).
```
This is leading to delayed in camera streams on our end.
kernel (ubuntu 18.04): 4.15.0-101-generic
version: 1.1.2.realsense-1.3.12
srcversion: BE1821F94C3D410077ACD10

Do you mind pointing me to the recommended mitigation step of how to patch the kernel?  Thanks!

MartyG-RealSense, 2022-04-11:
Hi @russellkir  Step 4 of the **Prerequisites** section of the RealSense SDK instructions for building from source code at the link below describe how to patch the kernel if you are building librealsense from source code.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites

For Ubuntu 18.04 / kernel 4.15, use the patch script **./scripts/patch-realsense-ubuntu-lts.sh**

*********

If you build librealsense from Debian packages then you do not need to patch the kernel as the patch is included in the packages.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

********

You also do not need to patch if you build librealsense from source code using the **libuvc backend / RSUSB backend** method, as such a build is not dependent on Linux versions or kernel versions and does not require patching.  It is a helpful build method for situations where a kernel conflict may be occurring, as this type of build bypasses the kernel.

The easiest way to build is with the **libuvc backend** method in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

*************

If you require a more customized build where you may need to configure various options or want to install compatibility 'wrappers' for platforms such as Python at the same time as librealsense, you can build with RSUSB support from source code using CMake by including the **-DFORCE_RSUSB_BACKEND=true** build flag in the CMake build instruction.  

Example instructions for such a build can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

russellkir, 2022-04-11:
@MartyG-RealSense we are installed from package as such:
```
apt-get install -y apt-transport-https rsync

# https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys C8B3A55A6F3EFCDE
add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
apt-get update

apt-get install -y --allow-unauthenticated \
    librealsense2 \
    librealsense2-dkms \
    librealsense2-utils \
    librealsense2-dev \
    librealsense2-dbg
apt-get clean
rm -rf /var/lib/apt/lists/*
```
If that includes the patch, can you provide guidance on still seeing the following issue? 
```[ 4694.898053] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 4694.988812] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 4695.307178] uvcvideo: Failed to query (GET_CUR) UVC control 2 on unit 3: -32 (exp. 1).
```
Thank you!

MartyG-RealSense, 2022-04-11:
There was a case with this error a couple of years ago where I consulted with Intel's RealSense team about it.  They provided the response below.

***********

The message “uvcvideo: Failed to query (GET_CUR) UVC control” indicates a failure on (kernel) driver level to read the requested parameter, for instance if the device was busy with internal operation or that the incoming requests queue is full.  In most cases this message can be disregarded as the SDK has built-in auto-retry mechanism to compensate for the temporalities.

MartyG-RealSense, 2022-04-11:
The possibility of implementing a mechanism to turn off debug messages to mute the warnings was considered by Intel but not proceeded with for the reasons provided below.

*********

Suppressing kernel-generated logs is not feasible by design. RealSense is a user-space SDK and must not affect third party applications (driver) running in the kernel space.

The kernel warning messages that arise from temporal glitches are cosmetic annoyances and there is not a degradation in the functionality of librealsense.

Depending on the use-case, the frequency that the messages are generated at may be between once every several seconds and a lengthy period of minutes between each one. The log may therefore typically not be a burden to read.

russellkir, 2022-04-11:
@MartyG-RealSense Thank you very much.  So the issue we run into is:
1. We attempt to start camera streams
2. We occasionally see a burst of those errors for ~40 or so seconds
3. It *does* eventually resolve, but leads to a delay in our recording program
So maybe I'd like to understand more about the following:

> that the incoming requests queue is full

Can you say more about this?

MartyG-RealSense, 2022-04-11:
If the computer is unable to keep up with processing the incoming flow of images then the 'frame queue' could become jammed full, resulting in errors.  You could try reducing frame rate (FPS) and / or resolution to see whether that reduces the frequency of the messages.

russellkir, 2022-04-11:
@MartyG-RealSense thanks!  We'll take what you've provided and come back if we have additional questions.  Thanks for the timely responses.

MartyG-RealSense, 2022-04-11:
You are very welcome to ask further questions if you have them.  Good luck!

wangshihui, 2022-11-01:
see here  [linux kenel log](https://linuxconfig.org/introduction-to-the-linux-kernel-log-levels#:~:text=When%20a%20log%20level%20is%20set%20as%20the,by%20messages%20about%20system%20instability%20or%20imminent%20crashes)
set the `console_loglevel` less   than 0 (KERN_EMERG) or  equals ,could prevent kernel  log message
eg :
below change is temporary only, will recover after reboot 
`echo -1 > /proc/sys/kernel/printk` 
or 
modify the  property `kernel.printk`   of  /etc/sysctl.conf  to -1 4 1 7 for permanent

> There was a case with this error a couple of years ago where I consulted with Intel's RealSense team about it. They provided the response below.
>
> The message “uvcvideo: Failed to query (GET_CUR) UVC [...]