realsenseai/librealsense #5304: Writing extrinsic calibration parameters using librealsense

nbenhaim, 2019-11-22:
Since the dynamic calibration tools/api are not available for Jetson Nano, I'm trying to figure out if there's a way to write the parameters after a third party calibration into the camera using just the librealsense API. (create something like the CustomRW tool)  According to the dynamic calibration documentation, it looks like the calibration API interfaces with librealsense, but I haven't been able to find out what function/api calls are available for this purpose.

Or better yet, if the calibration API can be open sourced or made available for the jetson nano / arm64 that would be ideal

thx

agrunnet, 2019-11-22:
Have you tried using the new self-calibration? It works on any platform and correct depth performance. It is in the RS viewer now and will be announced shortly with a white paper and api.

nbenhaim, 2019-11-23:
@agrunnet I wasn't aware of this - but I spent the day working with it and I'm a bit confused by how it works.  First, the on-chip calibration feature says it does some calibration simply by pointing the camera at a wall.  How is this possible?  Certainly it can't calibrate for depth unless given the ground truth?  It seems like this is what the "Tare Calibration" is for.  However, when I perform this Tare Calibration, the calibration parameters are not updated in the camera even after I get notified that camera was successfully updated.  I used CustomRW to put an incorrect translation value of -100 instead of the original -50, and after running Tare Calibration and the On-chip calibration, both do not update any of the camera's calibration params at all.  

Lastly, I am still interested in being able to write my own extrinsic params into the camera if possible, although if Tare Calibration can work I would be happy with that

thanks

agrunnet, 2019-11-23:
We will be releasing documentation shortly. 
The self-calibration adjust alignment to give best depth map. The Tare function refines it to give absolute accuracy and there needs user currently to input ground truth. This is not a full factory calibration so certain parameters are assumed to be unchanged, like baseline. If you adjust the “translation” extrinsic of distance between imagers then neither of these function will work. 
They both work by tweaking calibration by searching in nearby parameter space.

dorodnic, 2019-11-24:
Also, in the near-by release, we will be adding new APIs to support on-chip calibration, including API do write new calibration. For this release, it is supposed to be used with calibration you receive from on-chip method, but with some help from me and @agrunnet it should be easy enough to write custom values in code (you just need the table structure and the ability to recalculate table CRC)

nbenhaim, 2019-11-27:
@dorodnic @agrunnet  would be very useful to have some sample code or explanation on how to write back the extrinsic params between the stereo IR pair (L and R).  I see there is some table in the on chip calibration _new_calib and _old_calib and you write that data back so getting that format would be very helpful, as I have a time sensitive project I am working on.  Thanks!

dorodnic, 2019-12-01:
Such example will not make it into upcoming release (requires some time to prepare), but overall I think we are open to it in future releases (@agrunnet ?)
I can do my best with an explanation - 
All the different byte arrays you get from various new calibration APIs are holding the following structure: [ds5-private.h:337-349](https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-private.h#L337-L349)
If you modify any of the parameters you need to recalculate the `header.crc32`. Fortunately, there is `calc_crc` method in [types.cpp](https://github.com/IntelRealSense/librealsense/blob/9f99fa9a509555f85bffc15ce27531aaa6db6f7e/src/types.cpp#L588). At the moment these are not exposed in public APIs, but you should have no problem pulling them out to your application. Assuming CRC is correct, you should be able to set calibration to see how it looks and write calibration when you are satisfied. If you break something just call reset_to_factory_calibration.

nbenhaim, 2019-12-02:
thanks alot! I will let give it a shot

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

Hi @nbenhaim

There is still work being done on the on chip calibration process/documentation/updates but have you been able to test out the steps outlined by dorodnic?

I think we can close out this thread for now if nothing more is needed on the steps outlined.  And we will definitely not forget about the on chip calibration as it's something we are striving towards releasing documentation around it.

nbenhaim, 2020-01-07:
Actually yes! Sorry I didn't follow up but I was able to figure out how to
update the calibration table

closed 2020-01-23 by RealSenseCustomerSupport

rkelly-rtr, 2020-05-06:
Hi! I am also interested in writing custom calibration parameters to the realsense and the information and this information is very helpful. I wanted to confirm though that the calibration table only contains parameters for the two IR cameras and does not allow you to write calibration parameters to the rgb camera. @dorodnic Would you be able to confirm this? Thanks so much for the help in advance I really appreciate it.

> Such example will not make it into upcoming release (requires some time to prepare), but overall I think we are open to it in future releases (@agrunnet ?)
> I can do my best with an explanation -
> All [...]