realsenseai/librealsense #5307: Error in `python': free(): invalid pointer: 0x00000000901f5f68

likestudy, 2019-11-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300} | 
| Firmware Version                   | 2.29.0 | 
| Operating System & Version |  Linux (Ubuntu 16)| 
| Kernel Version (Linux Only)    |  4.4.0-166-generic| 
| Platform                                 | PC|

### Issue Description
When I want to use both pytorch and pyrealsense2, I get an error: * Error in `python': free(): invalid pointer: 0x00000000901f5f68 *.
The simple code I am running is:

import torch
import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30) 
pipeline.start(config)
pipeline.stop()

The problem disappears when I comment 'import torch' or 'pipeline.start(config)'.
It seems that there is a conflict between camera startup and the use of torch?

The details of the error are as follows：
_/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python /home/user/PycharmProjects/RealTimeGesRec/test_camera.py
*** Error in `/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python': free(): invalid pointer: 0x00000000028029c8 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f759a8867e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f759a88f37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f759a89353c]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZNSt8__detail9_CompilerINSt7__cxx1112regex_traitsIcEEEC1EPKcS6_RKSt6localeNSt15regex_constants18syntax_option_typeE+0x535)[0x7f7530ce1325]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZNK12librealsense8platform14v4l_uvc_device12get_profilesEv+0x6f6)[0x7f7530d1b386]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZNK12librealsense8platform26retry_controls_work_around12get_profilesEv+0x22)[0x7f7530d1bc72]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZN12librealsense10uvc_sensor20init_stream_profilesEv+0x1e2)[0x7f7530c3be12]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZN12librealsense12sr300_camera18sr300_depth_sensor20init_stream_profilesEv+0x54)[0x7f7530c8a384]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x54a84c)[0x7f7530c2f84c]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZNK12librealsense4lazyISt6vectorISt10shared_ptrINS_24stream_profile_interfaceEESaIS4_EEE7operateEv+0x5c)[0x7f7530c403ec]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZNK12librealsense11sensor_base19get_stream_profilesEi+0x52)[0x7f7530c31152]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZNK12librealsense4util6config11map_streamsEPKNS_16device_interfaceE+0x1d5)[0x7f7530bdbd75]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZN12librealsense4util6config7resolveEPNS_16device_interfaceE+0x44)[0x7f7530bdc6e4]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZN12librealsense8pipeline7profileC1ESt10shared_ptrINS_16device_interfaceEENS_4util6configERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x2c3)[0x7f7530bd9b23]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZN12librealsense8pipeline6config7resolveESt10shared_ptrINS_16device_interfaceEE+0x7ad)[0x7f7530bd4c9d]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZN12librealsense8pipeline6config7resolveESt10shared_ptrINS0_8pipelineEERKNSt6chrono8durationIlSt5ratioILl1ELl1000EEEE+0x31c)[0x7f7530bd528c]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZN12librealsense8pipeline8pipeline12unsafe_startESt10shared_ptrINS0_6configEE+0x910)[0x7f7530bd08d0]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(_ZN12librealsense8pipeline8pipeline5startESt10shared_ptrINS0_6configEES2_I18rs2_frame_callbackE+0xc5)[0x7f7530bd0d35]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(rs2_pipeline_start_with_config+0xa4)[0x7f7530ae0024]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x338665)[0x7f7530a1d665]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x3368a6)[0x7f7530a1b8a6]
/home/user/PycharmProjects/RealTimeGesRec/venv/lib/python3.5/site-packages/pyrealsense2/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x2f4f7f)[0x7f75309d9f7f]
/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python(PyCFunction_Call+0x77)[0x4ea117]
/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python(PyEval_EvalFrameEx+0x5a36)[0x529816]
/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python[0x52d2b9]
/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python(PyEval_EvalCode+0x1f)[0x52dfbf]
/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python[0x5fc652]
/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python(PyRun_FileExFlags+0x9a)[0x5feafa]
/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python(PyRun_SimpleFileExFlags+0x1bc)[0x5fecec]
/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python(Py_Main+0x456)[0x63ec96]
/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python(main+0xe1)[0x4d02e1]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f759a82f830]
/home/user/PycharmProjects/RealTimeGesRec/venv/bin/python(_start+0x29)[0x5d4b99]
======= Memory map: ========
00400000-007a8000 r-xp 00000000 08:0b 4340806                            /home/user/PycharmProjects/RealTimeGesRec/venv/bin/python
009a8000-009aa000 r--p 003a8000 08:0b 4340806                            /home/user/PycharmProjects/RealTimeGesRec/venv/bin/python
009aa000-00a41000 rw-p 003aa000 08:0b 4340806                            /home/user/PycharmProjects/RealTimeGesRec/venv/bin/python
00a41000-00a72000 rw-p 00000000 00:00 0 
01093000-0283f000 rw-p 00000000 00:00 0                                  [heap]
[... 276 lines cut ...]
7f759abd5000-7f759abd9000 rw-p 00000000 00:00 0 
7f759abd9000-7f759abf1000 r-xp 00000000 08:0a 2490574                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f759abf1000-7f759adf0000 ---p 00018000 08:0a 2490574                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f759adf0000-7f759adf1000 r--p 00017000 08:0a 2490574                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f759adf1000-7f759adf2000 rw-p 00018000 08:0a 2490574                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f759adf2000-7f759adf6000 rw-p 00000000 00:00 0 
7f759adf6000-7f759ae1c000 r-xp 00000000 08:0a 2490573                    /lib/x86_64-linux-gnu/ld-2.23.so
7f759ae53000-7f759afd9000 rw-p 00000000 00:00 0 
7f759aff2000-7f759aff3000 rw-p 00000000 00:00 0 
7f759aff3000-7f759b011000 r-xp 00000000 08:0a 2495181                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f759b011000-7f759b012000 r--p 0001d000 08:0a 2495181                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f759b012000-7f759b013000 rw-p 0001e000 08:0a 2495181                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f759b013000-7f759b014000 rwxp 00000000 00:00 0 
[... cut ...]
7f759b01c000-7f759b01d000 rw-p 00026000 08:0a 2490573                    /lib/x86_64-linux-gnu/ld-2.23.so
7f759b01d000-7f759b01e000 rw-p 00000000 00:00 0 
7ffe7682f000-7ffe76854000 rw-p 00000000 00:00 0                          [stack]
7ffe76931000-7ffe76933000 r--p 00000000 00:00 0                          [vvar]
7ffe76933000-7ffe76935000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)_

Any advices will help me a lot. Thanks.

asejam, 2019-12-03:
Hi @likestudy, I have the same issue. Were you able to resolve the problem?

likestudy, 2019-12-03:
This problem was solved as i **_import pyrealsense_** before **_import torch_**. 

The final code is:
_import pyrealsense2 as rs
import torch
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
pipeline.start(config)
pipeline.stop()_

I wonder if it can help you.

asejam, 2019-12-03:
Thank you very much, it worked!

closed 2019-12-03 by likestudy