realsenseai/librealsense #5308: Unable to set roi for auto_exposure

zheningy, 2019-11-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.12.00.00 | 
| Operating System & Version |  Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    | 4.15.0-60-generic                            | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.20.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  others                          | 

Hi, I met a problem that the auto exposure would over-expose center region while recording in dark environment. 
[image]
Thus I want to use roi auto-exposure to get normal brightness on this region like [this](https://github.com/IntelRealSense/librealsense/issues/4363).

However, I have issue with the  set_region_of_interest() function, which would crash for 
```
Exception caught : hwmon command 0x75 failed. Error type: Invalid parameter (-6).
```
Below is my code snippet.
 
```
  for (auto &&dev : ctx_.query_devices()) {
    if (rs_serial_number_.compare(
            dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER)) != 0) {
      continue;
    } else {
      for (auto &&sensor : dev.query_sensors()) {
        std::clog << dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) << std::endl;
        if (sensor.supports(RS2_OPTION_EMITTER_ENABLED)) {
          if (enable_laser_) {
            sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 1.f);
            std::clog << "Open realsense laser\n";
            continue;
          } else {
            sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 0.f);
            std::clog << "Close realsense laser\n";
            continue;
          }
        }
        if (sensor.supports(RS2_OPTION_ENABLE_AUTO_EXPOSURE)) {
          rs2::region_of_interest rs_roi;
          rs_roi.min_x = rs_auto_exposure_roi_.x;
          rs_roi.min_y = rs_auto_exposure_roi_.y;
          rs_roi.max_x = rs_auto_exposure_roi_.x + rs_auto_exposure_roi_.width - 1;
          rs_roi.max_y = rs_auto_exposure_roi_.y + rs_auto_exposure_roi_.height - 1;
          rs2::roi_sensor s(sensor);
          
          sensor.as<rs2::roi_sensor>().set_region_of_interest(rs_roi);
          std::clog << "rs_roi set to " << rs_auto_exposure_roi_ << std::endl;
          continue;
        }
      }
      cfg_.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
      break;
    }
  }

  pipe_ = rs2::pipeline(ctx_);
  if (enable_color_) {
    cfg_.enable_stream(RS2_STREAM_COLOR, color_width_, color_height_,
                       RS2_FORMAT_RGB8, color_fps_);
    std::clog << "rgb cam enabled: " << color_width_ << "x" << color_height_
              << " at " << color_fps_ << std::endl;
  }

  if (enable_depth_) {
    cfg_.enable_stream(RS2_STREAM_INFRARED, depth_width_, depth_height_,
                       RS2_FORMAT_Y8, depth_fps_);
    cfg_.enable_stream(RS2_STREAM_DEPTH, depth_width_, depth_height_,
                       RS2_FORMAT_Z16, depth_fps_);
    std::clog << "ir cam enabled: " << depth_width_ << "x" << depth_height_
              << " at " << depth_fps_ << std::endl;
  }

  pipe_.start(cfg_);
  std::clog << "pipe_ started" << std::endl;
  count_ = 0;

  // Capture 30 frames_ to give autoexposure, etc. a chance to settle
  for (auto i = 0; i < 30; ++i)
    pipe_.wait_for_frames();
```

It's appreciated if you could provide more examples about roi auto exposure!

ev-mp, 2019-12-01:
@zheningy hello,
The functionality of set/get_roi is available via extension API, and the way to check whether the active sensor supports this is by using:
```c++
if (sensor->is<roi_sensor>())
{
    sensor->as<roi_sensor>().set_region_of_interest(....
```
See the [implementation given]((https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp#L1887-L1892)) in realsense-viewer.

arpitg1304, 2019-12-12:
Is there a way to set the ROI in preset JSON file?

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

The ROI is not part of the JSON loader.
Thank you and please let us know if further assistance is needed.

zheningy, 2019-12-16:
After I used pdb, I found that the realsense will start four threads to start devices. Thus it is necessary to set a flag to avoid setting region of interest four times. Below is my code, could be useful.

`  for (auto &&dev : ctx_.query_devices()) {
    if (rs_serial_number_.compare(
            dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER)) != 0) {
      continue;
    } else {
      bool roi_flag = true;
      for (auto &&sensor : dev.query_sensors()) {
        if (sensor.supports(RS2_OPTION_EMITTER_ENABLED)) {
          if (enable_laser_) {
            sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 1.f);
            std::clog << "Open realsense laser\n";
          } else {
            sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 0.f);
            std::clog << "Close realsense laser\n";
          }
        }
        if (sensor.supports(RS2_OPTION_ENABLE_AUTO_EXPOSURE) & roi_flag &
            enable_roi_exposure_) {
          rs2::region_of_interest rs_roi;
          rs_roi.min_x = rs_auto_exposure_roi_.x;
          rs_roi.min_y = rs_auto_exposure_roi_.y;
          rs_roi.max_x =
              rs_auto_exposure_roi_.x + rs_auto_exposure_roi_.width - 1;
          rs_roi.max_y =
              rs_auto_exposure_roi_.y + rs_auto_exposure_roi_.height - 1;
          if (sensor.is<rs2::roi_sensor>()) {
            sensor.as<rs2::roi_sensor>().set_region_of_interest(rs_roi);
            roi_flag = false; // Avoid calling from multi threads
            std::clog << "rs_roi set to " << rs_auto_exposure_roi_ << std::endl;
          }
        }
      }
      cfg_.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
      break;
    }
  `

closed 2019-12-16 by zheningy