realsenseai/librealsense #5309: [D435 Not Working] Outdated UDEV-Rules after apt upgrade librealsense

ravijo, 2019-11-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 14.04.6 LTS  | 
| Kernel Version (Linux Only)    |  4.4.0-148-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Today, an update was available for the `librealsense` package, so I executed `apt upgrade` command. After the update, RealSense is throwing the following error-

```
Timestamp: 1574503699700.077637
Severity: Warn
Description: RealSense UDEV-Rules file:
 /etc/udev/rules.d/99-realsense-libusb.rules
 is not up-to date!
UDEV-Rules permissions configuration 
 for RealSense devices.

Missing/outdated UDEV-Rules will cause 'Permissions Denied' errors
unless the application is running under 'sudo' (not recommended)
In case of Debians use: 
sudo apt-get upgrade/install librealsense2-udev-rules
To manually install UDEV-Rules in terminal run:
$ sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger
```

I downloaded the latest file from here (Github) and executed following commands as suggested but nothing happened-
```
ravi@lab:~/Desktop$ ll
total 508
drwxr-xr-x  3 ravi ravi   4096 Nov 23 19:09 ./
drwxr-xr-x 43 ravi ravi   4096 Nov 23 19:11 ../
-rw-rw-r--  1 ravi ravi   5285 Nov 23 19:09 99-realsense-libusb.rules
ravi@lab:~/Desktop$ sudo rm /etc/udev/rules.d/99-realsense-libusb.rules
[sudo] password for ravi: 
ravi@lab:~/Desktop$ sudo cp 99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules
ravi@lab:~/Desktop$ sudo udevadm control --reload-rules
ravi@lab:~/Desktop$ udevadm trigger
```

**Please note that RealSense was working fine previously but not now.** 

Any workaround, please?

lostmsu, 2019-11-23:
Chirping in: I have the same problem with freshly installed librealsense on NVidia Xavier (arm64 Ubuntu 18.04)

ev-mp, 2019-11-26:
@ravijo hi, what SDK version are you running? 
Can you also clarify whether you're using Debian packages or building the SDK from source as those methods have separate installation methods and targets. 
In any event you should verify that a single udev-related file is installed, either with `sudo apt install librealsense2-udev-rules` or with `sudo cp ~/.99-realsense-libusb.rules /etc/udev.....`

ev-mp, 2019-11-26:
@lostmsu , please provide the problem description:
- SDK version used/ installation method
- Error message encountered

ravijo, 2019-11-27:
@ev-mp , thank you very much for the response.

> what SDK version are you running?

`librealsense` version shown by `relasense-viewer` is 2.28.1

> Can you also clarify whether you're using Debian packages or building the SDK from source.

I am using a prebuild package via `apt install`.

> you should verify that a single udev-related file is installed

How to verify it? I see a `99-realsense-libusb.rules` file inside `/etc/udev/rules.d/` directory.

_PS: I noticed that I can ignore this message. I can get the images from the camera surprisingly._

ev-mp, 2019-11-27:
@ravijo , I think it could be related to an issue with debian package that was fixed in v2.29, so I'd suggest to upgrade to v2.29/v2.30. (#4990)

>PS: I noticed that I can ignore this message. I can get the images from the camera surprisingly.

The message is informative to make the user aware of having multiple and potentially non-synchronized udev-rules. It does not affect the SDK functionality.
With that being said  - if you stick to Debian packages then make sure to delete `/etc/udev/rules.d/99-realsense-libusb.rules` as recommended.

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

Hi @ravijo,

Any update? Is your issue resolved per @ev-mp's suggestion?

ravijo, 2019-12-04:
Thanks, @RealSenseCustomerSupport, and @ev-mp for the support. It worked.

closed 2019-12-04 by ravijo