realsenseai/librealsense #5310: High CPU with rsusb backend

jamieforth, 2019-11-24:
| | |
|---|---|
|**librealsense**|2.30.0 RELEASE|
|**OS**|Debian testing (Linux kernel 5.2, unpatched)|
|**Name**|Intel RealSense D435I|
|**Firmware Version**|05.12.00.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|

CPU usages seems to be about 10 times higher when building the SDK using the new rsusb backend. Is that to be expected?

- `FORCE_RSUSB_BACKEND=true`
rs-depth: cpu ≈ 200% (i.e. maxing out two cores of a four core laptop) 

- `FORCE_RSUSB_BACKEND=false`
rs-depth: cpu ≈ 20%

dorodnic, 2019-11-25:
Hi @jamieforth 
Thanks for the feedback, we need to check.
My intuition is that it can have something to do with USB buffer management in user-space (as opposed to the relevant kernel module), but it is even more likely to be a bug (one of the thread is not managing its sleep time properly)

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Please let us know if this is still an issue with latest libRealsense build.
Thank you.

jamieforth, 2020-02-24:
Yes the issues still exists with the latest build.

The same comparison with `realsense-viewer`, depth stream alone is typically ~60% cpu, but when compiled with rsusb backend is ~120%.

Below are some messages printed to the console.

- `FORCE_RSUSB_BACKEND=true`

```
./realsense-viewer 
 24/02 23:05:24,801 WARNING [139794961086208] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed. Error type: No data to return (-21).
 24/02 23:05:27,539 ERROR [139794256484096] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/02 23:05:27,549 WARNING [139794961086208] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Enable / disable backlight compensation
 24/02 23:05:27,559 WARNING [139794961086208] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image brightness
 24/02 23:05:27,576 WARNING [139794961086208] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image contrast
 24/02 23:05:27,586 WARNING [139794961086208] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 24/02 23:05:27,596 WARNING [139794961086208] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image gamma setting
 24/02 23:05:27,607 WARNING [139794961086208] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image hue
 24/02 23:05:27,617 WARNING [139794961086208] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image saturation setting
 24/02 23:05:27,627 WARNING [139794961086208] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image sharpness setting
 24/02 23:05:27,637 WARNING [139794961086208] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 24/02 23:05:27,647 WARNING [139794961086208] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Enable / disable auto-exposure
 24/02 23:05:27,657 WARNING [139794961086208] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Enable / disable auto-white-balance
 24/02 23:05:32,460 WARNING [139794717853440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/02 23:05:32,761 WARNING [139794717853440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/02 23:05:33,063 WARNING [139794717853440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/02 23:05:33,367 WARNING [139794717853440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```

- `FORCE_RSUSB_BACKEND=false`

```
$ ./realsense-viewer 
 24/02 23:04:43,465 ERROR [140266267678464] (types.h:304) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid or incomplete multibyte or wide character
 24/02 23:04:43,465 WARNING [140266267678464] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 24/02 23:04:43,471 ERROR [140266267678464] (types.h:304) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid or incomplete multibyte or wide character
 24/02 23:04:43,471 WARNING [140266267678464] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 24/02 23:04:59,082 WARNING [140266343663424] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed. Error type: No data to return (-21).
 24/02 23:04:59,258 WARNING [140266343663424] (backend-v4l2.cpp:1208) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
```

SirDifferential, 2020-02-25:
This is an issue on Raspberry Pi 4 due to the low end nature of the SOC. When the RSUSB backend ends up using so much CPU, less is available for other tasks. This relates to other problems as well such as #5412

RealSenseCustomerSupport, 2020-02-27:
----------------------------------------------

@SirDifferential, thank you so much for the input.

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

Please let us know if there is any other question or if it is alright to close this ticket.
Thank you.

SirDifferential, 2020-03-04:
@RealSenseCustomerSupport This problem still exists. Do not close this ticket.

SirDifferential, 2020-03-27:
The latest development branch with f01419932588e4b4011964a5b1da968b2f61ab69 in it seems to have this issue solved. On Raspberry PI 4 librealsense with `-DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=Release` has these CPU usage numbers:

Current master: 270% CPU
Current development: 83% CPU

RealSenseCustomerSupport, 2020-04-01:
----------------------------------------------

Thank you, @SirDifferential. This is also addressed in the latest SDK build v.2.34.0.
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#bug-fixes-and-improvements

closed 2020-04-01 by RealSenseCustomerSupport

jayv, 2020-06-05:
With a D435 I'm seeing 100% CPU (single core) for an async pipeline 1920x1080 RGB8 on an i7-9750H Ubuntu 20.04 Linux 5.4 with librealsense built via vcpkg (statically). 
The vcpkg version is at 2.34, but I've also changed the portsfile to 2.35 (with and without FORCE_RSUSB) and always get the same 100% CPU usage. Interestingly when linking (dynamically) against the ubuntu `.deb` packages on the same machine, I'm only observing ~15% CPU usage for the same workload. 
Linux Perf revealed that in the 100% CPU case, 93% was spent doing `librealsense::unpack_yuy2` (doing a bunch of `mm*` calls), where as in the low CPU usage case only 58% was spent in that method, but there are no `mm*` calls!? 

Perhaps @dorodnic has insights?

sharpriy, 2022-08-11:
Hi @dorodnic , 
D415 is connected to Centos7 system and i don't want to do kernel patch.
when i execute any of example ( like rs-hellorealsense). It gives this error

[error] RealSensor:  RealSense error calling rs2_pipeline_start ( pipe:0xd8f960 ) The requested pixel format 'Z16 ' is not natively supported by the running Linux kernel and likely requires a patch Last Error: Invalid argument

It doesn't support Z16 format (depth)

Is there any other way to communicate with camera?