realsenseai/librealsense #5313: RuntimeError: Profile does not contain the requested stream

raishafok, 2019-11-24:
Traceback (most recent call last):
  File "t265_stereo.py", line 145, in <module>
    streams = {"left"  : profiles.get_stream(rs.stream.fisheye, 1),
RuntimeError: Profile does not contain the requested stream

RealSenseCustomerSupport, 2020-01-02:
----------------------------------------------

Hi Raisha,

Can you provide some more context here? We are not seeing this error.

Thanks

closed 2020-01-15 by RealSenseCustomerSupport

GOBish, 2020-06-29:
I get the same error when trying to run t265_stereo.py.

```
Traceback (most recent call last):
File "t265_stereo.py", line 145, in
streams = {"left" : profiles.get_stream(rs.stream.fisheye, 1),
RuntimeError: Profile does not contain the requested stream
```

t265_example.py, t265_rpy.py both work well.

GOBish, 2020-06-29:
@raishafok did you have a second realsense camera plugged in when you tried to run this example?
I got the same error when a D435 and T265 were both connected, but when I removed the D435 and left the T265 connected and ran t265_stereo.py, the program ran.