realsenseai/librealsense #5314: x,y,z coordinates

gioannides, 2019-11-24:
| | |
|---|---|
|**librealsense**|2.30.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D415|
|**Serial Number**|851112060689|
|**Firmware Version**|05.11.15.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|844513020425|
|**Firmware Update Id**|844513020425|

Please provide a description of the problem

For some reason, the camera sometimes just reports (x, y, z) = (0, 0, 0) after passing the RGB and D coordinates (which are all within the frame coordinates) to the function 
`x, y, z = rs.rs2_deproject_pixel_to_point(intr,[int(avg_x),int(avg_y)],float(avg_z))`.
Intrincics are obtained by doing:
`   intr = profile.get_stream(rs.stream.depth) `
   ` intr = intr.as_video_stream_profile().get_intrinsics()`

ev-mp, 2019-11-26:
Hi @gioannides, unfortunately the description is not sufficient to get insights:
- Why you're referring to `RGB and D` - does it mean that you're using aligned frames? If so - how those are generated.
- Mathematically it is expected to get (0,0,0) as output when `D=0`. Is there a check that `D` passed into the function is a valid (non-zero, non-negative) value ?

Please provide a more comprehensive code segment/reproduction flow for review

ev-mp, 2019-12-01:
@gioannides , did you manage to resolve it? 
Please update

gioannides, 2019-12-02:
Hello, given a depth picture, how can I obtain non-zero and non-negative pixels. For example maybe obtaining neighbouring pixels if one is faulty?

ev-mp, 2019-12-09:
This probably should be dealt with on a case-by-case basis as the requirements are very situation and context-dependent.
Perhaps the more legitimate question is how to minimize the number of non-valid pixels in the original depth image, for which I would recommend to start with the [BKM](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf) white paper

RealSenseCustomerSupport, 2019-12-17:
----------------------------------------------

@gioannides Any other questions with the reply of ev-mp? Looking forward to your update. Thanks!

ev-mp, 2020-01-06:
No response received

closed 2020-01-06 by ev-mp