realsenseai/librealsense #5315: "No frames received" due to user permissions

iban-rodriguez, 2019-11-25:
|   |  |
| ------------- | ------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version | Linux (Ubuntu 18.4) | 
| Kernel Version (Linux Only)    |  4.9.140 | 
| Platform                                 | NVIDIA Jetson (nano and TX2)  |
| SDK Version                            | 2.30.0 }                          | 
| Language                            |  C, Python | 
| Segment			|   | 

### Issue Description

Using information at:
https://github.com/jetsonhacks/buildLibrealsense2Xavier
https://devtalk.nvidia.com/default/topic/1051016/jetson-nano/use-intel-reg-realsense-trade-d435-series-depth-camera-on-nano/

we have been able to compile and make D435 work correctly on Jetson TX2i. Surprisingly, executing the exact same steps on another TX2 and on Jetson Nano we found that:
* rs-depth: Works properly and receive frames from the camera
* rs-color: Open the camera correctly but fails with "No frames received" message.
* python: When configuring the camera to use only depth stream, frames are received but when configuring the camera to use RGB stream (or both), script hangs when calling pipeline.wait_for_frames() and fails reporting "No frames received".

After lot of testing, recompiling, etc. we luckily found that the problem was not related to compilation nor to the hardware. The problem was that user we used to test the camera on second TX2 and Nano **does not belong to video group**. Adding the user to video group solves the problem. The issue does not occur on PC as my user does not belong to video group but is able to read RGB frames.

It is very confusing to be able to open the camera and request frames from it when the user do not have permissions to do it and just report "No frames received". Library should report some kind of error more related to the real problem when requesting for an operation that the user is not allowed.

dorodnic, 2019-11-26:
Hi @iban-rodriguez 
Usually, it does (Library should report some kind of error more related to the real problem)
Thank you for the report, I need to understand what exactly went wrong. 
The way it supposed to work is udev rules setup should handle the permissions for relevant product IDs, and in case of missing permissions you should see something along the lines of access denied error message (and also a warning in the Viewer). 

Have you tried the recently released debian packages for the Jetson family? 
(If I understand correctly the method you linked is based on patching the kernel and building the SDK from source using the V4L2 backend. We know this process has a lot of moving parts and is very error prone, this is why Debian packages since version 2.26 are using the libuvc backend, not requiring kernel patching or proper setup of V4L2 specific permissions)

iban-rodriguez, 2019-11-27:
Hi @dorodnic 

Yes, we tried the Debian packages 2-3 weeks ago more or less and also compiled the last version of the library forcing libuvc, and we decided to go on patching the kernel and using V4L2 backend because we found other issues we couldn't overcome:
- CPU usage is greater for the same channels/fps requested when using libuvc instead of V4L2. We tested it on PC and verified after on Jetson when V4L2 worked.
- Impossible to get 30 FPS on Jetson TX2 when requesting RGB, depth and IR streams simultaneously. CPU goes quite up and resulting FPS never goes higher than 22.
- Camera open failure when tried to open it more than twice. This is the most serious issue as it requires unplugging and plugging camera again to make it usable

All these issues have disappeared using V4L2 backend, being the only problem the reported in this ticket.

ev-mp, 2019-12-01:
@iban-rodriguez, we do use both TX2 and Nano boards for development but afair this is the first report on user/group-specific issue with either Tegra or Non-Tegra environments, so we'd like to receive more details on the initial user's permissions that led to the inconsistency.
In general the  `udev-rules` script is intended to handle device permissions, so it is weird that it seem to allow for RGB but not Depth sensor.
Can you confirm that the `udev-rulles` script that you use is not corrupted and that the issue is reproducible with the script properly installed?

ev-mp, 2019-12-09:
@iban-rodriguez hello, do you have an update/confirmation for the original report ?

iban-rodriguez, 2019-12-10:
@ev-mp, I have checked the installed udev file and its contents is:
```
# Intel RealSense UVC cameras (R200, F200, SR300 LR200, ZR300, D400, L500)
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0a80", MODE:="0666", GROUP:="plugdev", RUN+="/usr/local/bin/usb-R200-in_udev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0a66", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0aa5", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0abf", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0acb", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad0", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="04b4", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad1", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad2", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad3", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad4", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad5", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad6", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0af2", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0af6", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0afe", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0aff", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b00", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b01", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b03", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b07", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b0c", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b0d", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b3a", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b3d", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b48", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b49", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b4b", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b4d", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ab3", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0adb", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b55", MODE:="0666", GROUP:="plugdev"
SUBSYSTEMS=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="8087", ATTRS{idProduct}=="0af3", MODE="0666", GROUP="plugdev"
SUBSYSTEMS=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="8087", ATTRS{idProduct}=="0b37", MODE="0666", GROUP="plugdev"
SUBSYSTEMS=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="03e7", ATTRS{idProduct}=="2150", MODE="0666", GROUP="plugdev"
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad5", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
DRIVER=="hid_sensor_custom", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad5", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p && chmod 0777 /dev/%k'"
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0af2", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
DRIVER=="hid_sensor*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0af2", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p && chmod 0777 /dev/%k'"
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0afe", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
DRIVER=="hid_sensor_custom", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0afe", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p && chmod 0777 /dev/%k'"
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0aff", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
DRIVER=="hid_sensor_custom", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0aff", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p && chmod 0777 /dev/%k'"
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b00", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
DRIVER=="hid_sensor_custom", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b00", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p && chmod 0777 /dev/%k'"
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b01", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
[... cut ...]
```
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense D435
    Serial number: 909512072304
    Firmware version: 05.11.01.100

rs_error was raised when calling rs2_pipeline_wait_for_frames(pipe:0x5573548020):
    Frame didn't arrived within 15000
```

If I just add the user to `video` group, `rs-color` works correctly.

In the other hand, `rs-depth` demo application always works correctly.

We are using a custom rootfs, not the official one. Maybe a package is missing?

ev-mp, 2019-12-10:
@iban-rodriguez , in the above `udev-rule` file the registration is added to the `plugdev` group, and this is actually the group that most Linux users are automatically assigned.
I tried to run `groups` shell command on several PCs including JetsonTX2 and Raspbian and for all environments the user included `plugdev` affiliation.
I haven't experimented with `video`,  but at least on some PCs `video` tag was not listed and it didn't prevent cameras from streaming properly. Additionally  I'm not sure that adding 'video' alone will support hot plug in functionality.

II would recommend to check if assigning the `plugdev` group to all users by default does not break your organization policies/security arrangements. Otherwise you'll probably have to build and maintain a customized udev-rules file for the rootfs.

RealSenseCustomerSupport, 2020-01-09:
----------------------------------------------

Hi @iban-rodriguez,

Did you have a chance to check @ev-mp's advice out? Please let us know if you need any further assistance with this matter.

Thank you!

closed 2020-01-21 by RealSenseCustomerSupport