realsenseai/librealsense #5316: Problems reading playback of T265 with only the pose stream

AJMena, 2019-11-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10}   | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.25.0.993 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
I'm currently reading files using this method:
https://gist.github.com/MatsumotoJ/e23a80f540b58db12ee92d8a96de1745

Everything works fine when my playbacks have all the streams. When the file of the T265 only contains the pose stream reading the playback sometimes almost blocks. When in the "readframe" function "playback.pause();" is called sometimes takes 10 seconds. When reading normally is instantly. In the end to read and synchronize a frame takes minutes when normally takes less than a second. This happens sometimes and is completely random.

Any idea on how to solve this?

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

Hi AJMena,

We are unable to reproduce this problem. Using the sample code below I was able to successfully and quickly load and read frames out of a T265 bag file. Can you provide the bag file that you're using that *only* contains the Pose data as you stated above.

Thanks

AJMena, 2019-12-09:
[Intel RealSense T265.zip](https://github.com/IntelRealSense/librealsense/files/3939198/Intel.RealSense.T265.zip)

I leave a file that have the issue just in case.
But for my part I changed the code to not do pause/resume and avoid the blockings. So now the code works fine and faster.

```
// readframe: read the next frame of a bag-file-playback-pipeline
int readframe(rs2::pipeline & pipe, size_t & framenumber, rs2::frameset & frameset)
{
	auto device = pipe.get_active_profile().get_device();
	rs2::playback playback = device.as<rs2::playback>();

	frameset = pipe.wait_for_frames();

	if (frameset[0].get_frame_number() < framenumber)
	{
		return EOF;
	}

	framenumber = frameset[0].get_frame_number();

	return 0;
}

// readframeto: read the frame just after the specified timestamp.  
int readframeto(double timestamp, rs2::pipeline & pipe, size_t & framenumber, rs2::frameset & frameset)
{
	while (true)
	{
		if (timestamp <= frameset.get_timestamp())
		{
			return 0;
		}

		frameset = pipe.wait_for_frames();

		if (frameset[0].get_frame_number() < framenumber)
		{
			return EOF;
		}

		framenumber = frameset[0].get_frame_number();
	}

	return 0;
}
```

Reading without pause/resume to close the file(s) correctly we must start "while (app)" like this:

```
while (app)
{
	// We must close it all correctly or may block
	if (flag_end == true)
	{
		int i = 0;
		for (auto &&pipe : pipes) // loop over pipelines
		{
			readframetoEOF(pipes[i], framenumbers[i], framesets[i]);
			pipe.stop();
			i++;
		}
		app.close();
		continue;
	}
```

It is necessary to read the files completely to close it correctly. For that purpose, we have the readframetoEOF function:

```
// readframetoEOF: read frames until the EOF.  
int readframetoEOF(rs2::pipeline & pipe, size_t & framenumber, rs2::frameset & frameset)
{
	if (frameset[0].get_frame_number() < framenumber)
	{
		return EOF;
	}
	while (true)
	{
		frameset = pipe.wait_for_frames();

		if (frameset[0].get_frame_number() < framenumber)
		{
			return EOF;
		}

		framenumber = frameset[0].get_frame_number();
	}

	return 0;
}
```

RealSenseCustomerSupport, 2019-12-09:
----------------------------------------------

Thanks AJ,

We will investigate with bag file. Thanks for providing. So are you saying with the new code it is no longer a problem for you and we can close this specific issue... correct?

AJMena, 2019-12-10:
That's correct. Thank you for the support.

closed 2019-12-10 by AJMena