realsenseai/librealsense #5317: Include the version of the pyrealsense

pedrombmachado, 2019-11-25:
----------------------------------------------------------------------------------------------------
| Required Info                   |   Include __version__ variable with the pyrealsense version                                      |
|---------------------------------|------------------------------------------- |
| Camera Model                       | All | 
| Firmware Version                   | All | 
| Operating System & Version |   All  | 
| Kernel Version (Linux Only)    | Alll                                         | 
| Platform                                 | Alll  |
| SDK Version                            | Latest and future versions}                          | 
| Language                            | python                          | 
| Segment			|  All                        | 

### Issue Description
Dear pyrelsense2 developer team,
   Can you please add the variable __version__ with the version of pyrealsense2 version? 
```
import pyrealsense2 as rs
>>> rs.__version__
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute '__version__'

```
  This simple addition would make our life much easier when working with embedded devices.
Many thanks in advance,
Pedro

dorodnic, 2019-11-25:
Yes, we should do this. @lramati

bfulkers-i, 2019-11-27:
See #5337

closed 2019-11-28 by pedrombmachado