realsenseai/librealsense #5318: Error installing D435 Drivers using SDK on windwos 7 (Error code 267)

Katarina-Poole, 2019-11-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   Win 7 | 
| Platform                                 | PC |
| SDK Version                            | 2.0 Win7                        | 

### Issue Description
I'm having issues with installing the drivers for the camera on a PC with a Windows 7 operating system. The following errors get thrown:

Unable to execute file: C:\Windows\system32\pnputil.exe
CreateProcess failed; code 267.
The directory name is invalid.

The Windows 7 OS is up to date, I've also run the installation file as an administrator, and when running 'pnputil /?' in the windows command window the dir of 'C:\Windows\system32' it sees the executable. 

Because it throws this error I am unable to install any of the drivers so the camera doesn't show up in the viewer. 

I was wondering if there is a solution to this?

dorodnic, 2019-11-25:
What version of the SDK are you installing?

Katarina-Poole, 2019-11-25:
Apologies I should of been more specific, the version is Intel RealSense SDK 2.0 (Win7) 2.29.0.1122

dorodnic, 2019-11-26:
The reason I asked this is because I think we solved this issue in 2.30 :)
Let me know if the new release is suitable to your needs.

Katarina-Poole, 2019-11-26:
That worked a treat, thank you very much!

closed 2019-11-26 by Katarina-Poole