realsenseai/librealsense #5319: Wrong timestamp obtained in D435i

fanofpal, 2019-11-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 5.11.15.0| 
| Operating System & Version |  Ubuntu 18  | 
| Kernel Version (Linux Only)    | 5.0.0-36-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                          | 
| Language                            |  python                        | 
| Segment			|  others                          | 

### Issue Description
Hello, I was tring to collect some data used for a SLAM project. However, the frame timestamp obtained from frame seemed to have some problems. The following are the printed timestamps of consecutive frames. I used the code below to get them.
`timestamp = frames.get_timestamp()`
...
timestamp: 1574686990183.428
timestamp: 1574686990217.0005
timestamp: 1574686990217.0005
timestamp: 3160492351536.7637
timestamp: 3158446977537.991
timestamp: 3158446977571.5474
timestamp: 3156898314297.376
...
[image]

Then I checked the timestamp in Realsnse-viewer, it showed that the frame timestamps(global time) were also wrong (something like 3148958470379) in color and depth stream, although that in IMU streams were correct.
[image]

argosvr, 2019-11-27:
did you try patching your kernel ?

fanofpal, 2019-11-28:
> did you try patching your kernel ?

Yes, I installed the SDK following the [Linux manual installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md), the patched kernel modules were also applied successfully.
Actually, every time I reopen the color and depth streams, they can get the normal global timestamp--but only for a few seconds, then they suddenly change to wrong number, and never return to normal--unless I reopen them again.
However, the IMU stream is always normal.

JeremyBYU, 2019-11-30:
Same issue as #4505 . Read my comments there to understand problem. This problem is reproducable on 2 x86 computers and a raspberry pi for all parched/unpatched kernels from 5.0-5.3.

The issue goes away for me using 4.18 kernel using dkms install.

fanofpal, 2019-12-04:
@JeremyBYU Thanks for your reply, I encountered the same problem as you, and now I switched back to the 4.15 kernel.

RealSenseCustomerSupport, 2020-01-02:
----------------------------------------------

Hi @fanofpal

Based on your previous comment, you're able to get things working.  Is there anything more necessary on this thread?

ev-mp, 2020-02-04:
The resolution was successfully integrated into `development` branch. scheduled for the upcoming release.

RealSenseSupport, 2020-05-15:
Hi @fanofpal 

Have you tried the latest librealsense?  There has been quite a bit of PRs that have been put in place recently that should fix your issue.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-05-22 by RealSenseSupport