realsenseai/librealsense #5320: Realsense conveyor belt application

ArjanVroegop, 2019-11-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (5.11.15.0| 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  (4.15.0  (with UVC patch)                                     | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.30}                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
Hi everyone,

I'm looking into the possibilities to use a Realsense (D435) for a conveyor application. So we have a conveyor belt with an encoder with a more or less constant speed (the speed can vary, I still have to gather more info about this). The RealSense will be used to inspect an object, and if the object has to be processed a robot arm will pick up the object. To classify the object, both RGB and depth information are taken into account.

I've seen many topics about timing and timestamps, but I couldn't find an answer to this question:

Even when the hardware sync is enabled (to sync RGB and depth), it seems that depth and RGB frames are not grabbed at the same moment (they run the the same interval, but not in sync). This can be seen in the depth and RGB images, where a shift in the direction of travel can be seen between both images. Is there a way to take both images at the same time, or at least minimize the time in between? Is there any documentation that describes when the images are taken exactly?

Also, for this application a frame-rate of about 2 fps is sufficient. The lowest possible settings is 6 fps, but it seems that the hardware sync does not work (I can't get RGB and depth framerate equal in the realsense viewer, for 15 and 30 fps it works with the right settings). Is this a known issue or am I doing something wrong?

dorodnic, 2019-11-26:
Hi @ArjanVroegop 
One approach you could check-out is using IR information instead of RGB. IR frames are always 100% synchronized with depth, temporally and spatially. 
When using the RGB sensor, there is always going to be small differences in sensor exposure times and configuration, but the difference between the two should never exceed one frame. This means that you could run both streams at 60 fps, sample the SDK at the rate you need (say twice a second) and get up to 16 ms delay between the images (I think it would be actually up to 8 ms, but I'm not 100% sure)

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

Hi @ArjanVroegop

Let  us know if there is anything further needed after the outline from dorodnic.  If not then we can close out this thread.

closed 2020-01-23 by RealSenseCustomerSupport