realsenseai/librealsense #5321: librealsense CMake fails config when added as a git submodule

JTrantow, 2019-11-25:
I am trying to update the RealSense examples of the Open3D project. https://github.com/intel-isl/Open3D to work with both the old cameras (SDK 1) and the newer cameras (SDK 2).

The Open3D project contains a CPP example for the older realsense cameras and old source code for librealsense SDK 1 to build a library. The library would only build if I added a path to the system includes to find <librealsense\rs.h>. I can replace the librealsense library source code with a git submodule of the v1.12.4 tag (Open3d/3rdparty/librealsense) and the original example now builds correctly for the BUILD_LIBREALSENSE. **SDK 1 as a submodule was easier than expected!!!**

I tested the Open3D RealSense.cpp example with a R200 camera and I can get the depth stream visualization to work correctly if I disable the color stream. If I include the color stream I'm hitting some exception. I'm not sure what's wrong with the example but it does achieve the goal of linking the camera CPP example to SDK 1 as submodule so I'll consider SDK 1 good enough for now.

But I am having problems adding librealsense SDK 2 as a submodule to Open3D...

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R200 / D400  | 
| Firmware Version                   | R200  1.0.72.10, D400 05.12.00.00 | 
| Operating System & Version |   Win10| 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  SDK 1 v1.12.4, SDK 2 2.30.0 (head) | 
| Language                            |  C++                          | 
| Segment			|  {other }                          | 

### Issue Description
Adding librealsense to the Open3D project as a 3rdparty submodule does not build under Windows. I have not tried the other build platforms. I think the problem in CMake configure indicates a CMake directory problem will carry over to the other submodule builds. 

Specifically, librealsense is included into another project (Open3D) as submodule, CMAKE_SOURCE_DIR is set to the Open3D directory and not the directory setup by the librealsense submodule. 

I'm not an CMake expert with big projects. I modified the Open3d/3rdparty/CMakelist.txt with: (this was enough to get the SDK tag to work with Open3D)
Note: I use 3rdparty/librealsense for SDK 1 and librealsense2 for SDK2 and only install one SDK at a time.

# RealSense SDK2
if (BUILD_LIBREALSENSE2)
    message(STATUS "Building LIBREALSENSE2 from source")
    add_subdirectory(librealsense2)
    Directories("${CMAKE_CURRENT_SOURCE_DIR}/librealsense2" librealsense_INCLUDE_DIRS)
endif ()

When CMake runs config, it gives errors because it is using CMAKE_SOURCE_DIR as the open3d location rather than the open3d/3rdparty/librealsense2. If I insert the location of the librealsense submodule "/3rdparty/librealsense2/" after CMAKE_SOURCE_DIR into the following CMakelists.txt's I can get through the Configure step.

include(${CMAKE_SOURCE_DIR}/CMake/opengl_config.cmake) - src/gl/CMakeLists.txt line 36
include(${CMAKE_SOURCE_DIR}/3rdparty/librealsense2/CMake/opengl_config.cmake) - tools/CML.txt line 22

1. Should something in the Open3D CMake files be changing CMAKE_SOURCE_DIR to the 3rdparty/librealsense2 directory when building librealsense?
2. Or should librealsense CMakeList.txt's be using a relative path so it builds correctly as a submodule?
3. Or should Open3D CMake be building a smaller subset of librealsense when it just needs the library?  (Is there a way for the Open3D CMakeFiles to disable librealsense BUILD_GRAPHICAL_EXAMPLES, BUILD_EXAMPLES, etc.. ???)

CMake config output:

Building LIBREALSENSE2 from source
Internet connection identified
Info: REALSENSE_VERSION_STRING=2.30.0
Setting Windows configurations
using RS2_USE_WMF_BACKEND
CMake Error at 3rdparty/librealsense2/src/gl/CMakeLists.txt:36 (include):
  include could not find load file:

    C:/Users/JJT/Documents/MTSI/Open3D/CMake/opengl_config.cmake

CMake Error at 3rdparty/librealsense2/examples/CMakeLists.txt:14 (include):
  include could not find load file:

    C:/Users/JJT/Documents/MTSI/Open3D/CMake/opengl_config.cmake

Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
Unable to find apriltag library, skipping pose-apriltag example
CMake Error at 3rdparty/librealsense2/tools/CMakeLists.txt:22 (include):
  include could not find load file:

    C:/Users/JJT/Documents/MTSI/Open3D/CMake/opengl_config.cmake

Building with TM2
Download TM2 firmware 0;"No error" for target-0.2.0.857.mvcmd
Download TM2 firmware 0;"No error" for central_app-2.0.19.271.bin
Download TM2 firmware 0;"No error" for central_bl-1.0.1.112.bin
libusb not found; using internal version (missing: LIBUSB_LIB LIBUSB_INC) 
Fetching recommended firmwares:
D4XX_RC_VERSION: 5.11.15.0
D4XX_FW_VERSION: 5.11.15.0
SR3XX_FW_VERSION: 3.26.1.0
Download firmware 0;"No error" for D4XX_FW_Image-5.11.15.0.bin
Download firmware 0;"No error" for D4XX_RC_Image-5.11.15.0.bin
Download firmware 0;"No error" for SR3XX_FW_Image-3.26.1.0.bin

JTrantow, 2019-11-25:
I just realized CMake GUI does display the options for both Open3D and the librealsense submodule. So I can disable the librealsense examples and get past the CMake Configure and Generate errors. But now when I compile within Visual Studio 15 2017 Win64, I'm getting thousands of compile errors. If I go back into CMake and disable the BUILD_LIBREALSENSE2 option (disables the librealsense submodule) I get a relatively clean Open3D build?

Checking the BUILD_LIBREALSENSE2 option should only enable building the RealSense2.cpp project and the librealsense library. The compile errors show up all over the place (other projects besides the RealSense example or library). Lots of C2039 'strlen': is not a member of "global namespace" and identifier not found errors from iosfwd, cstring, xlocale, etc indicating something in librealsense submodule process is messing up the rest of the projects. (changing include directories???)

Summary:

1. I can build Librealsense by itself without any problems.
2. I can build Open3D by itself without any problems.
3. If I add librealsense as a submodule to Open3D AND disable the librealsense BUILD_EXAMPLES option, I can CMake Configure/Generate without errors. 
4. BUT librealsense submodule build is messing something up because the VisualStudio compile is getting thousands of errors because it isn't looking in the correct places???
6. If I disable the CMake options that include librealsense,  Open3D again builds without any problems.

Any solutions or hints would be appreciated as I'm stuck.

dorodnic, 2019-11-26:
Hi @JTrantow 
Thank you for the feedback, this is a great initiative and I'm looking forward to see it come together.
That said, I have never been a fan of building the SDK from within other projects. 
What do you think about the approach of building the SDK with CMake, running install (either `sudo make install` or install target in VS) and then using CMake find commands to add it to Open3D?
After-all you rarely build opencv (or other sizable projects) as part of another project, for example. 
With SDK 2.0 there is some added complexity coming from recording capabilities, tools and wrappers that were absent from librealsense1. While this all should be solvable, the sub-module approach sounds like the harsher road.
Another approach we have considered is to flip this around and add an Open3D example to the SDK.

JTrantow, 2019-11-26:
I'd be fine with any method that works. I opened an initial Open3D pull
request with a RealSense SDK 2 example where I added the SDK 2 include
files and requested direction on how to add the complete SDK 2 that would
work with other builds. The response was to use submodules
https://github.com/intel-isl/Open3D/pull/1329 which worked for updating
with the latest SDK 1 tag.

I'm a rookie with CMake but I think some of the 3rdparty CMakelist.txt
files have options for building from source or finding installed versions.
So I think CMake find would be fine. Any pointers on CMake find with
librealsense?

Adding an Open3D example to librealsense would also work and might be a
better fit for my needs.  I just started a project where I know I want
librealsense, opencv, and possibly open3D available. I would like a clean
starting point so I can easily merge upstream improvements into primarily
Windows but possibly Unix platforms.

JTrantow, 2019-11-26:
P.S. I think the problem with RealSense as a submodule to Open3D is a compiler setting other than the include directories (which are the same). Looks more like an incompatible C++ language or _STD setting.

The 3rdparty/CMakeLists.txt has a few find_package() and pkg_search_modules(). I will look into using that to add librealsense.

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

@JTrantow Welcome your contribution! Did you create any PR for this now? Thanks!

JTrantow, 2020-01-30:
Gave up on Open3D.

closed 2020-01-30 by JTrantow