realsenseai/librealsense #5323: Is there a Python version of this code available?

RajkiranVeldur, 2019-11-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | D435i 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | Ubuntu 18.04
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 5.0.0-36-generic
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  | PC
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | NA
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | Python 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | others

### Face Antispoofing in Python
Can someone help me to conver the below chunk of code to Python. Appreciate your help.

bool find_depth_from(
    rs2::depth_frame const & frame,
    float const depth_scale,
    dlib::full_object_detection const & face,
    markup_68 markup_from, markup_68 markup_to,
    float * p_average_depth
)
{
    uint16_t const * data = reinterpret_cast<uint16_t const *>(frame.get_data());
    float average_depth = 0;
    size_t n_points = 0;
    for( int i = markup_from; i <= markup_to; ++i )
    {
        auto pt = face.part( i );
        auto depth_in_pixels = *(data + pt.y() * frame.get_width() + pt.x());
        if( !depth_in_pixels )
            continue;
        average_depth += depth_in_pixels * depth_scale;
        ++n_points;
    }
    if( !n_points )
        return false;
    if( p_average_depth )
        *p_average_depth = average_depth / n_points;
    return true;
}

RajkiranVeldur, 2019-11-27:
Can someone tell me which pixel this line refers to?

auto depth_in_pixels = *(data + pt.y() * frame.get_width() + pt.x());

maloel, 2019-11-27:
@RajkiranVeldur the depth frame data is stored in units that are called "pixels" here -- there's no standard convention that I know of, so that's the name that was used. Each "pixel" needs to then be scaled to an actual depth (in meters) by using the depth_scale.

RajkiranVeldur, 2019-11-28:
@maloel thanks for the response. Does this mean that the variable will return the depth value of the landmark associated with it? For e.g., will it return **x-depth** value of **nose_landmark**?

maloel, 2019-11-28:
@RajkiranVeldur The variable 'depth_in_pixels' is the depth (not in meters) of that specific landmark, yes.
The function iterates thru the landmark indices you give it and averages their depth.

RajkiranVeldur, 2019-11-28:
@maloel thanks for replying. We were trying to implement this algorithm using MTCNN instead of Dlib, yet it wasn't accurate. May I know, if you people have tried with any other face detection algorithm?

maloel, 2019-11-28:
@RajkiranVeldur There's a pull request waiting right now that implements OpenVINO face detection (though without landmarks, which can be fairly easily added). I find the OpenVINO implementation ([model](https://docs.openvinotoolkit.org/latest/_models_intel_face_detection_adas_0001_description_face_detection_adas_0001.html)) more robust (finds faces in a wider range of positions).

Check out their model zoo [here](https://software.intel.com/en-us/openvino-toolkit/documentation/pretrained-models).

RajkiranVeldur, 2019-11-28:
@maloel OpenVINO is great, however we are more aligned towards MTCNN implementation for facial landmark detection but the accuracy is not even close to Dlib. Our main concern is we are planning to implement it at the edge. So, we are looking for a lightweight model and MTCNN is already lightweight.

Can you please guide us, how we can improve our accuracy with MTCNN based spoof-detection model?

maloel, 2019-11-28:
Hi @RajkiranVeldur,

Can you please clarify what you mean by the accuracy issues? Are you referring to the accuracy of the depth you get from realsense, or the accuracy of MTCNN itself (such that the 5 points are off)?

Assuming the latter, it sounds like you need something more along the lines of DLIB's 68-point model which would provide many more points of reference. I did find one [project](https://itnext.io/face-api-js-javascript-api-for-face-recognition-in-the-browser-with-tensorflow-js-bcc2a6c4cf07) with lightweight 68-point landmark extraction for javascript. Maybe the models it uses can be of use to you?

RajkiranVeldur, 2019-11-30:
Hi @maloel yes, I was referring to the accuracy of using MTCNN with Realsense. Most of the time it was showing even a real person as Spoof. Might be, as you said it is because MTCNN gives us very fewer landmarks.  

Also, may I get your's or anyone's email from the Realsense team? Actually, we would like to use Realsense with our Facial recognition product. We are from Techolution India Pvt Ltd and are working on a D435i camera for spoof detection and have NDA with Intel for information request regarding the RealSense (NDA Number: cnda037223). If it is not possible to share here, can you please email me on rajkiran.veldur@techolution.com

Thank you.

dorodnic, 2019-12-02:
Hi @RajkiranVeldur 
You can send me an email to sergey.dorodnicov@intel.com
Any general questions about using the technology are better discussed here, so more people could benefit from it

RajkiranVeldur, 2019-12-02:
Hi @dorodnic thank you for sharing the contact details. Can you please help me with the C++ code to display some text on the frame window, depending on the condition?

maloel, 2019-12-02:
@RajkiranVeldur 
Putting text on a DLIB frame is a bit unintuitive.
Check [here](https://stackoverflow.com/questions/29751319/print-text-in-dlib-window). Let me know if it works :).

RajkiranVeldur, 2019-12-07:
@dorodnic @maloel can you tell me what are the landmarks you people are considering for the facial features?

> @RajkiranVeldur
> Putting text on a DLIB frame is a bit unintuitive.
> Check [here](https://stackoverflow.com/questions/29751319/print-text-in-dlib-window). Let me know if it works :).

Hey, @maloel it worked. Thank you :)

RajkiranVeldur, 2019-12-07:
@dorodnic @maloel can you please tell me what are the landmarks you are considering for the facial features? Ear, Nose, Eye, Chin and Mouth with respect to the image below.

[image]

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi @RajkiranVeldur

Is there anything further necessary on this thread?

***If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by RealSenseCustomerSupport