realsenseai/librealsense #5324: in realsense viewer

wangqihong, 2019-11-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17)  } | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
The displayed device error is particularly large, the actual distance is 2.4m, and the measured distance is 3m. The closer the error is, the closer the error is 0.
<Describe your issue / question / feature request / etc..>

g2-bernotas, 2019-11-26:
May you need to recalibrate the camera? I never needed to be done before, but your experienced error is indeed large. Check [#2329](https://github.com/IntelRealSense/librealsense/issues/2329) problem.

wangqihong, 2019-11-27:
[](url) @g2-bernotas  Does the depth map calibration separately calibrate the left and right grayscale images?

wangqihong, 2019-11-27:
@Can these calibration APIs calibrate the d435i, why do I show no device connection?

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

Please double check version of the tool. D435i support was added in version 2.6.8.
https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API

closed 2019-12-11 by RealSenseCustomerSupport