realsenseai/librealsense #5327: How can I make monochrome Depth image from D435i

Dennis2Choi, 2019-11-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)}                                              | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC                                                          |
| SDK Version                            |  { legacy / 2.2.29 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {**Robot**/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
 - How can I make monochrome Depth image from D435i ?
 - It automatically generate colorful depth image from recorder. 
 - Is there any setting to change it (code change)
   Or is there any way to change it from the output png file to monochrome png data?

 Thanks in advance

dorodnic, 2019-11-27:
Hi @Dennis2Choi 
Converting depth data to image is done via `colorizer` class. 
In RealSense Viewer, you can configure it using Depth Visualisation panel. 
In code, you can set option `RS2_OPTION_COLOR_SCHEME` to change how depth appears.

Dennis2Choi, 2019-12-03:
Thanks anyway
I think I can try using below closed issues 
https://github.com/IntelRealSense/librealsense/issues/1831
https://github.com/IntelRealSense/librealsense/issues/2785
https://github.com/IntelRealSense/librealsense/issues/3168

closed 2019-12-03 by Dennis2Choi