realsenseai/librealsense #5329: RuntimeError

Dontla, 2019-11-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi, I encountered a very strange problem. I connected six cameras to the device. I used the `len (ctx.devices)` function to print the number of cameras. It was displayed as 6, but when I accessed the camera's serial number and usb port, it went wrong on the sixth camera. I don’t know why, so I want to ask you for help.
best regard!
dontla
`        ctx = rs.context()`
`        # Check if the cameras are all connected`
`        cam_num = len(ctx.devices)`
`        print(cam_num)`
`        if cam_num < 6:`
`            print('The cameras are not all connected!')`
`        else:`
`            # Print the serial number of the connected camera and its USB interface number`
`            n = 0`
`            for i in ctx.devices:`
`                n += 1`
`                print('serial number {}：{}；usb port：{}'.format(n, i.get_info(rs.camera_info.serial_number),
                                                               i.get_info(rs.camera_info.usb_type_descriptor)))`
`                print(i.get_info(rs.camera_info.serial_number))`
[image]
[image]

ev-mp, 2019-12-01:
@Dontla hello, can you provide kernel and librealsense logs when this error occurs for review ?
I also noticed that while mentioning SDK v2.25 the attached screenshot shows a different version.
What is the situation when you enumerate devices with Viewer v2.25 ?

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@Dontla Could you please provide more information as ev-mp indicated? Thanks!

Dontla, 2019-12-13:
I guess the camera is unstable at startup, just like the engine needs to run for a while before the car starts in winter, so I use len (context.query_devices ()) to detect the number of cameras. When the number of cameras is equal to I expect the value to remain the same, which means that the camera is stable. At this time, I go to get the camera serial number and other information. No error has occurred for the time being.

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@Dontla Glad to hear the problem resolved. Any other questions about this? Thanks!

closed 2019-12-20 by RealSenseCustomerSupport