realsenseai/librealsense #5330: How to calculate rotation & translation matrix of each image (non-checker board ) from

Shivani1796, 2019-11-27:
I am using T265 RealSense camera from that I have taken 100 images (left+right camera) and want to generate trajectory from the given images for which I need rotation and translation matrix of each image . I am moving my camera and taking images so for each image(left+right) will generate one rotation and translation matrix so how can I generate for 100 images ? Is there any specific code which can help me generate the matrix .

Shivani1796, 2020-03-06:
Thanks for the reply @RealSenseCustomerSupport yes, you understood my question in a right way.
Can I know what are the exact intrinsic calibration parameter of Intel T265 camera, this is just for the verification.

RealSenseCustomerSupport, 2020-03-06:
----------------------------------------------

Hi @Shivani1796

There is information in the SDK that provides intrinsics.  Below is a python example of getting and using intrinsics.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py

Let me know if you need anything else on this topic.

Shivani1796, 2020-03-12:
Hiello,
Thanks to the help been solved that issue.
I want to do Odometry with Intel D435i camera is it possible? Can you tell
me intrinsic camera parameters of D435i camera ?
Regards
Shivani Baldwa
Ming Chi University Of Technology
- 84 Gungjuan Rd, Taishan Dist, New Taipei City 24301
-
-
-
-
-
-
-
-  Taiwan

RealSenseSupport, 2020-04-30:
Hi @Shivani1796 

There are functions to get intrinsic values for the D435i camera within the librealsense library, a simple search will allow you to get that information.  

The T265 topic was covered by the link in comments above.  I think this thread is solved now.

closed 2020-05-29 by RealSenseSupport