realsenseai/librealsense #5332: T265 covariance issue

ogil0018, 2019-11-27:
We tested the T265 on a forklift truck. It is mounted on top view direction to the ceiling. The position data is recorded. We use ROS Melodic on a Linux Ubuntu 18.04. system (without Realsense viewer). We tested two hardware configurations (Odroid XU4 with librealsense 2.30.0 and Pokini F2 with librealsense 2.29.0).

The symptom is that during the recording the covariance jumps suddenly (from 0.01 to 10.0) and the device stops sending position data (“nan” instead). This can be seen in the sent rosbag logdata files (requested by Intel). This event happened on the truck roughly about 30 to 60 minutes after start but happened even after 4 hours. It does not(!) happen when the sensor is stationary, e.g. in an office/desk environment.

-	How much memory can the internal (feature) map consume, recorded by the T265?
-	Is it possible to save the T265 internal map data after a recording?
-	Is it possible to save position data even though the covariance is >= 10.0?

dorodnic, 2019-12-01:
Hi @ogil0018 
I believe the team has done significant work to combat NaN issue
Unfortunately, T265 doesn't know how to provide proper covariance of pose data. ROS wrapper is generating fake covariance matrix from T265 confidence measure.
@bfulkers-i - do you know the answer to map memory consumption?

radfordi, 2019-12-06:
@ogil0018, the memory consumed by the T265 map is small (a few megabytes) because we limit the map size.   Because of the fixed size, we are unlikely to run out of memory.  FWIW, we recently doubled the map size, and we are experimenting with further increases.

You can export the current map with `rs2_export_localization_map`, and then load it in during subsequent runs.  The map can be preserved around `stop/start` by setting the new option `RS2_OPTION_ENABLE_MAP_PRESERVATION`, so you can avoid having to save and load.  See #5208.  This was the default in 2.29.0.   In either case, in order to make use of the loaded/preserved map you needed to have called `rs2_set_static_node`.  See #5356 for an example/demo.

The covariance you are getting is an artifact added by the ROS wrapper.  `librealsense` only provides a confidence "enum".   This accounts for the discrete jump.  The discrete value would just tell you that there is no valid pose.   You can use a loaded/preserved map along with a static node to get the pose of the static node in the coordinates of new T265 session.

A year ago we had memory fragmentation issues after running for many hours, but that should be fixed now.  We also found the leaving the device sitting still (with motion in the scene?) would lead to NaNs.  We recently got to the point that that this was hard to repeat without shaking the device vigorously.   Your recording may help us make further progress.

neilyoung, 2019-12-06:
> The map can be preserved around `stop/start` by setting the new option `RS2_OPTION_ENABLE_MAP_PRESERVATION`, so you can avoid having to save and load. See #5208. This was the default in 2.29.0.

@radfordi  Jim, if you say, MAP_PRESERVATION was default in 2.29.0, does it mean, it is no longer default? And does that correlate with my observations here?  https://github.com/IntelRealSense/librealsense/issues/5343
Sorry for jumping in.

radfordi, 2019-12-06:
@neilyoung, it was the default in 2.29.0 and is no longer. I suspect you haven't makde it far enough down in your email to see my https://github.com/IntelRealSense/librealsense/issues/5343#issuecomment-562363468 there.  :)

neilyoung, 2019-12-06:
@radfordi Arghh, sorry. Indeed, I haven't seen this. Thanks for reminding me and keep on doing your great job. Will close #5343

RealSenseSupport, 2020-09-15:
Thank you for highlighting the unfortunate continuation of the NaN issue while using T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-21 by RealSenseSupport