realsenseai/librealsense #5338: Intel realsense viewer and upgrade tool cannot talk to D435i

jff8, 2019-11-28:
| | |
|---|---|
|**librealsense**|2.29.0 RELEASE|
|**OS**|Windows|

Intel realsense viewer and upgrade tool cannot talk to D435i.
Ugrade tool is unable to find any cameras.
Windows device manger shows camera correctly
When running RealSense viewer I get error: Backend in rs2_create_device (info_list:11c7ceb8,Index:0) MFCreate Device Source(_device_attrs, &_source) returned HResult 0x800700

MartyG-RealSense, 2019-11-28:
I hope that the link below will be of help, as it describes a similar case to yours.

https://github.com/IntelRealSense/librealsense/issues/3381

Can you check please that your Windows privacy settings allow access to camera applications.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

jff8, 2019-11-28:
Thanks yes it was similar to #3381 , permissions issues.  Please close .

closed 2019-11-28 by jff8