realsenseai/librealsense #5339: Frame rate is very slow in dlib face detection

subbuweb, 2019-11-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |  ( 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.30 }                          | 
| Language                            |  {C#/C++ }                          | 
| Segment			|  {Face Anti spoof }                          | 

### Issue Description
i trying with dlib face detection example. it's working but frame rate is very slow.. how to resolve. also i need c# equivalent code for face anti spoof detection.

MartyG-RealSense, 2019-11-28:
Face detection tends to be processing intensive.  It may help to add hardware acceleration to your project with an Intel Neural Compute Stick 2 USB device alongside your RealSense camera.  An example of the results achievable with the stick can be seen in the YouTube video in the link below.

https://youtube.com/watch?v=wF3JD3jyGaA

subbuweb, 2019-11-29:
Please provide sample code for face anti spoof (liveness) detection in c#. we have using intel realsense D435 camera

subbuweb, 2019-12-09:
please give me some solutions..

MartyG-RealSense, 2019-12-09:
If you absolutely need to use C# in your anti spoofing solution, the best solution I could find right now was a commercial solution called Cyblerlink FaceMe which supports integration into the user's products via C#.

https://www.cyberlink.com/stat/technology/enu/tech_face.jsp

Please scroll down to near to the bottom of the page to find the section headed 'Secure Anti-Spoofing', which has an accompanying demonstration video.

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

@MartyG-RealSense Thanks very much for the share!
@subbuweb Did you get chance to try and any other questions about this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@subbuweb Could you please update? If we don’t hear from you in 7 days, this issue will be closed. Thanks!

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@subbuweb Sorry we didn't hear from you for weeks. Will close the ticket at this point. Please feel free to create another ticket if you still have issue or questions. Thanks!

closed 2020-02-17 by RealSenseCustomerSupport