realsenseai/librealsense #5340: Setting Advanced Mode Settings (Depth Table Control) in Android

ssakhavi, 2019-11-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17) / MacOS  | 
| Platform                                 | Android |
| SDK Version                            |  { 2.30.0 }                          | 
| Segment			|  {Smartphone}                          | 

### Issue Description

In advanced mode, there are a few parameters that can be set so we can get closer to objects which we are trying to measure using the realsense camera. 

I've noticed that they are not in the current Android codebase.

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

@ssakhavi Could you please specify the detailed parameters you're looking for? Thanks!

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

@ssakhavi  Could you please specify more details so that we can further look at this? Thanks!

ssakhavi, 2019-12-23:
Sorry for the late reply. 

We need to :

- Modify the Depth Control Table in order to set the distance in which the device can pick up the 3D measurements.
- We also need to get the intrinsic parameters to do the 3D measurement as well.

I've already implemented it in our own code.

RealSenseCustomerSupport, 2020-05-18:
----------------------------------------------

@ssakhavi Sorry for late response. Have submitted your request to our engineering team. Any update will let you know. Thanks!

RealSenseSupport, 2021-11-04:
@ssakhavi Any update from your side? Please note that the ticket will be closed if we don't hear from you for another 7 days. Thanks!

ssakhavi, 2021-11-04:
Hi
Unfortunately, I'm not working with the sensor anymore.
Thank you for responding though.

RealSenseSupport, 2021-11-08:
@ssakhavi Thanks for your update! We will close this ticket accordingly.

closed 2021-11-08 by RealSenseSupport