realsenseai/librealsense #5341: Can RealSense SR300 work for Raspberry pi 4?

calvin8899wan, 2019-11-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Raspbian Buster (September 2019)  | 
| Kernel Version (Linux Only)    |  ( Linux kernel 4.19.75)                                         | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            |  2.30.1                         | 
| Language                            |  C++                        | 
| Segment			|                      | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Raspberry pi has two USB3.0 port , so I use the USB3.0 port to connect the Realsense SR300 , but the Raspberry Pi didn't detect the SR300.  I'm not sure SR300 is support to the Raspberry pi4? I didn't see any SR300 for Rpi4 in Internet. And i don't know where can find the Firmware Version.
  
I also use the other method to check the Realsense SR300, but didnt work.
pi@raspberrypi:~ $ rs-enumerate-devices
No device detected. Is it plugged in?

This below is the command i use to compile th c++ code  & the console output
pi@raspberrypi:~/Desktop $ sudo g++ test.cpp -o xxx -I/usr/include/opencv -lopencv_core -lopencv_highgui  -lopencv_imgcodecs -lopencv_imgproc -lopencv_video -lopencv_flann -lopencv_videoio -lGL -lGLU -lglut

/usr/bin/ld: /tmp/ccwQyI2x.o: in function `rs2::error::error(rs2_error*)':
test.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x1c): undefined reference to `rs2_get_error_message'
/usr/bin/ld: test.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x68): undefined reference to `rs2_get_failed_function'
/usr/bin/ld: test.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x80): undefined reference to `rs2_get_failed_function'
/usr/bin/ld: test.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0xf8): undefined reference to `rs2_get_failed_args'
/usr/bin/ld: test.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x110): undefined reference to `rs2_get_failed_args'
/usr/bin/ld: test.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x180): undefined reference to `rs2_get_librealsense_exception_type'
/usr/bin/ld: test.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x194): undefined reference to `rs2_free_error'
/usr/bin/ld: /tmp/ccwQyI2x.o: in function `rs2::error::handle(rs2_error*)':
test.cpp:(.text._ZN3rs25error6handleEP9rs2_error[_ZN3rs25error6handleEP9rs2_error]+0x20): undefined reference to `rs2_get_librealsense_exception_type'
/usr/bin/ld: /tmp/ccwQyI2x.o: in function `rs2::points::get_vertices() const':
test.cpp:(.text._ZNK3rs26points12get_verticesEv[_ZNK3rs26points12get_verticesEv]+0x34): undefined reference to `rs2_get_frame_vertices'
/usr/bin/ld: /tmp/ccwQyI2x.o: in function `rs2::points::get_texture_coordinates() const':
test.cpp:(.text._ZNK3rs26points23get_texture_coordinatesEv[_ZNK3rs26points23get_texture_coordinatesEv]+0x34): undefined reference to `rs2_get_frame_texture_coordinates'
collect2: error: ld returned 1 exit status

ev-mp, 2019-12-01:
@calvin8899wan hello,
Recently there was [some work](https://github.com/IntelRealSense/librealsense/pull/5025) done to fix RPI4 build specific issues. And it was verified to work with D400/SR300/T265 correctly.

Things to check:
>I also use the other method to check the Realsense SR300, but didnt work.
pi@raspberrypi:~ $ rs-enumerate-devices
No device detected. Is it plugged in?

Make sure to install the `udev-rules` according to the installation guide: ```./scripts/setup_udev_rules.sh```.
Another thing to mention is that all the tests we did were with `rsusb` and not `v4l` backend in order to allow for IMU sensors in addition to Video. And I don't have a first-hand report of `v4l` feasibility, though I wouldn't expect problems with SR300.
In case you build default  (v4l) configuration and still experience issues , then you can experiment with USB backend by setting `cmake .. ... -DFORCE_RSUSB_BACKEND=true` for SDK v2.30+ or `-DFORCE_LIBUVC` for the previous versions.

>sudo g++ test.cpp -o xxx -I/usr/include/opencv -lopencv_core -lopencv_highgui -lopencv_imgcodecs....

The SDK's library seem not appear in the linker's list  -  
>-lopencv_highgui -lopencv_imgcodec **-lrealsense2** ...

ev-mp, 2019-12-09:
@calvin8899wan hello, did you manage to install and run the setup?
Do you need more assistance with it ?
Please update

alexazorn, 2019-12-11:
Hi, i'm working with both D415 and SR305 on Raspberry Pi 4 and works well. Be sure that you configure the librealsense correctly, see the raspbian installation guide and it should work. https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

calvin8899wan, 2019-12-14:
Sorry,it's my problem,after i reinstall the setup and it finally can work. thanks a lot.

closed 2019-12-16 by RealSenseCustomerSupport