realsenseai/librealsense #5342: T265 not found in Unreal Engine 4 Wrapper

codewing, 2019-11-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.857 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  -                                       | 
| Platform                                 | PC |
| SDK Version                            |  2.30.0                          | 
| Language                            | Unreal Engine 4.22                         | 
| Segment			|  others                         | 

### Issue Description
I've connected the T265 to my computer. Checked whether it works in the Viewer and then tried to find it in the template Unreal Engine project.
The code compiles successfully but when starting the demo level it states no device found and also the refresh in the inspector doesn't show the device.

[image]

codewing, 2019-11-29:
After failing to use the device in UE4 I've tried to use it directly from the c++ sdk and to check whether it's an issue specifically related to the UE4 code.

I've successfully compiled everything but when I run ``rs-enumerate-devices`` or the ``rs-fw-update`` tool no devices show up.

[image]

radfordi, 2019-12-05:
This looks like a permissions issue.  We're having trouble opening the device.

codewing, 2019-12-09:
Seems like starting it once with administrator rights made it work.

closed 2019-12-09 by codewing